T-vK/ESP32-BLE-Keyboard #2: Works with iPad Air2, iPad Pro 2, iPhone SE. Does NOT work with iPad 2

satoer, 2019-08-20:
I can confirm the example code works with iOS 12.3.1 on:
iPad Air 2: Bluetooth 4.2
iPad Pro 2: Bluetooth 5.0
iPhone SE: Bluetooth 4.2

It does not work with iOS 9.3.5 on an iPad 2 (Bluetooth 2.1). Sadly that’s the device I want to use it for. I want to use it as domotica dashboard and wake it up remotely by using Bluetooth commands. It does not show up in the Bluetooth devices when I try to pair it. Other Bluetooth keyboards work fine. Is the a Bluetooth 2.1 limitation for this to work, and is this solvable?

T-vK, 2019-08-20:
This is probably because the iPad 2 doesn't support Bluetooth LE. I don't think this can be solved easily. It would probably require a whole new set of libraries.
Your other keyboards probably just work because they use the old non-LE protocol.

satoer, 2019-08-20:
Okay, was afraid of that. Anyway, you can "Check" the "Compatible with iOS (Untested)" in the notes.
But I would set a footnote: *iPad 4 and higher, iPhone 4S and higher. (These support BLE)

T-vK, 2019-08-21:
I'll see if I can find the time to update it later today. Thanks for testing btw!

T-vK, 2019-08-27:
I don't know, but my best guess is that the iPhone 8 Plus has a different Bluetooth chip and either the chip, its firmware or its driver don't (fully) understand Bluetooth LE. Maybe you can find more information on the matter by googleing for "iPhone HOGP support" or something like that. I think HOGP (HID over GATT Profile) is what needs to be supported properly for BLE keyboards and mice to be recognized properly.

T-vK, 2019-08-28:
That wouldn't make much sense, as for me it works on either 5.0 (on my Android phone) and 4.2 (on my Linux PC). And for satoer as well on various Apple devices (see [post 1](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/2#issue-482741158)).
I'd say it's more likely the device runs a different chip/driver/firmware.

satoer, 2019-08-28:
I've tested sucessfully with an iPad Pro 2, which is also Bluetooth 5.0

MrSwiss, 2019-11-10:
I had tried on lastest iPad air. Keyboard works, but unfortunately I can get set batterylevel to work. I can only set initial battery level but can't up date it.

T-vK, 2019-11-11:
Yes, I think that doesn't work on any device for now. I'm not exactly sure why it's not working, but I haven't really looked at it.

blackketter, 2020-02-06:
The example works great with my iPhone 8, iOS 13.3.1.

xundre, 2020-06-05:
This was my first ESP32 project, and unfortunately has been an epic fail as I wanted to use it with and old Ipad2 . 

Has any one had any luck with this?

blackketter, 2020-06-05:
The ESP32 hardware only supports Bluetooth LE, which was introduced with Bluetooth 4.0.   Older devices that only support older versions of Bluetooth (like iPad 2 or iPhone 4) can’t be added. 

This issue should probably be closed.

T-vK, 2020-06-06:
Well, it's an ongoing issue. If I close it, I suspect someone else will open a new one. 

But to add to your comment and to clarify some things in general for everyone here, just because a device supports Bluetooth 4.0, it doesn't mean that it supports Bluetooth LE, as it's an optional part of the specification. My understanding is, a Bluetooth 4.0 device merely has to support one of these:  Classic Bluetooth, Bluetooth High Speed or Bluetooth Low Energy.

And then there is the problem that just because a device says it supports Bluetooth Classic or Bluetooth LE, it doesn't mean that it supports every single profile.
Looking at the [Bluetooth API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/index.html) you can see that the ESP32 generally supports Classic and LE, but it only supports a handful of Classic profiles, HID not being part of them. For LE however you can see it supports the GATT profile which can be used for a BLE keyboard using the [attributes HOGP or HIDS](https://www.bluetooth.com/specifications/gatt/).

So if an iPad/iPhone doesn't fully support the GATT profile (which is a Bluetooth LE only thing) or at least the HOGP or HIDS attributes of it (not sure which one is used by the BLE HID library under the hood), then it is currently not possible to connect the ESP32 to those devices for this purpose. 
And if an iPad/iPhone supports HOGP or HIDS is dependent on chip, firmware, driver and the kernel's and operating system's API.

T-vK, 2020-10-21:
Hey everyone, I am trying to make a list of apple devices that work or don't work with this library.
If you read this and you have an Apple device that you have tested, I would very much appreciate it if you could add a comment in #51. (Unless of course it is already listed in there.)

T-vK, 2021-08-30:
#111 Includes a bunch of fixes that should improve iOS compatibility. We still need some people to test it though.