T-vK/ESP32-BLE-Keyboard #3: adding additional media keys?

msmithkey, 2019-10-21:
Just found your project after beating my head against the wall trying to learn the ESP-IDF. This is the only project I have found that seems to do security in a way where iOS will accept inputs and I have set up a simple program to send  single characters with  pushbuttons and it works great. thanks for the hard work. One little annoyance of iOS and bluetooth keyboards is that they hide the on screen keyboard whenever a bluetooth keyboard is connected. I am trying to make an ebook page turner remote for a relative who has limited mobility and would like to read on her phone on a gooseneck mount while in bed. In order to toggle the OSK when a BT keyboard is connected, i need have the BT device send the eject key. I have done this in the past using the HID-project.h library
I can't remeber which one worked but it was either
[HID_CONSUMER_EJECT](https://github.com/NicoHood/HID/blob/d4938ddcff7970bc1d32a040a08afeac4915e4a9/src/HID-APIs/ConsumerAPI.h#L154)
or
[HID_CONSUMER_STOP_SLASH_EJECT](https://github.com/NicoHood/HID/blob/d4938ddcff7970bc1d32a040a08afeac4915e4a9/src/HID-APIs/ConsumerAPI.h#L174)
I am very new to C and kind of out of my depth in trying to implement this and was hoping someone could point me in the right direction. Thanks again for the great project. I am already dreaming up a way to augment the lack of a physical play/pause button in my honda  when using carplay and other interesting little projects.

T-vK, 2019-10-21:
You could try to replace
https://github.com/T-vK/ESP32-BLE-Keyboard/blob/63e08b72f2a2c5ffae00950cb17450e4986ee768/BleKeyboard.cpp#L76
with
```
  USAGE(1),           0xB8,          //   USAGE (Eject)                ; bit 4: 16
```

And also replace
https://github.com/T-vK/ESP32-BLE-Keyboard/blob/63e08b72f2a2c5ffae00950cb17450e4986ee768/BleKeyboard.h#L67
with 
```
const MediaKeyReport KEY_MEDIA_EJECT = {16, 0};
```

Then you should be able to use:

```
bleKeyboard.write(KEY_MEDIA_EJECT);
```

msmithkey, 2019-10-21:
Awesome thanks! will do. I will test this out when i get home tonight. I'm planning on open sourcing it, even though it will be pretty simplistic. Right now I'm just doing proof of concepts and learning new platforms. Appreciate the help

DhruvGupta5, 2020-05-30:
> KEY_MEDIA_EJECT

Hi i have two questions 

> You could try to replace
> 
> https://github.com/T-vK/ESP32-BLE-Keyboard/blob/63e08b72f2a2c5ffae00950cb17450e4986ee768/BleKeyboard.cpp#L76
> 
> with
> 
> ```
>   USAGE(1),           0xB8,          //   USAGE (Eject)                ; bit 4: 16
> ```
> 
> And also replace
> 
> https://github.com/T-vK/ESP32-BLE-Keyboard/blob/63e08b72f2a2c5ffae00950cb17450e4986ee768/BleKeyboard.h#L67
> 
> with
> 
> ```
> const MediaKeyReport KEY_MEDIA_EJECT = {16, 0};
> ```
> 
> Then you should be able to use:
> 
> ```
> bleKeyboard.write(KEY_MEDIA_EJECT);
> ```
> 
> If you think your project could help other people, consider creating a repository for it here on GitHub.

hi i am new to cpp, 
can you kindly answer my two questions
1. whats the role of bit in the following line ?
USAGE(1),           0xE2,          //   USAGE (Mute)                ; bit 4: 16 (copied from https://github.com/T-vK/ESP32-BLE-Keyboard/blob/07e4bda48229400b7a55b440a9e0591ef306447f/BleKeyboard.h#L67 )
2. in reference to the value of HID_CONSUMER_EJECT = 0xB8 from
https://github.com/NicoHood/HID/blob/d4938ddcff7970bc1d32a040a08afeac4915e4a9/src/HID-APIs/ConsumerAPI.h#L154
do we still have to use KEY_MEDIA_EJECT for this library ?

paulruvolo, 2023-03-08:
I tested T-vK's suggested modification to support the eject key, and it works!  The only caveat (and maybe this is obvious to others) is that you should unpair with the keyboard and then repair to make sure the device you are connecting to is using the new mapping.

AlexSmirnov9107, 2024-09-30:
@paulruvolo 
could you give a example? i've tested. my esp32 crashed

RabByte-0, 2025-01-11:
Sorry to add to the noise here but I'm new and trying to work something out. 
I'm trying to add two macOS specifics commands [0x02A0] and [0x029F]. does it matter what media key I overwrite and is there anything specific I need to to do since it is a uint16_t instead of uint8_t?