T-vK/ESP32-BLE-Keyboard #4: KEY_MEDIA_VOLUME_UP (and other media keys) don't seem to work on my Android Device

speendo, 2019-11-12:
First of all, thank you very much for your work on this project.
Also, I have to apologize, that my issue might not be too insightful - I am just starting with Bluetooth...

I was able to connect my ESP32 with my Android device. However, I tried several media keys (e.g. `KEY_MEDIA_VOLUME_UP` and `KEY_MEDIA_PLAY_PAUSE` however, they have not been recognized (or at least, the Android device showed no reaction).

What could be the reason for that?

T-vK, 2019-11-12:
What is your Android version?
Do the other keys work on your Androidi device?
Do the media keys work when you try it with a different device (e.g. a Windows or Apple device)?

speendo, 2019-11-12:
Sorry, I didn't have the chance to test most of that, as I don't have too many bluetooth devices available atm.

I am using Android 9 and I can confirm that characters and the `Enter` key are working as expected.

speendo, 2019-11-13:
Sorry, it seems I didn't try hard enough yesterday. Today I was able to make it work.

Sorry for the inconvenience!

T-vK, 2019-11-14:
Neat. Do you know what caused your issue? (Just in case someone else gets the same problem in the future.)

speendo, 2019-11-14:
Thank you for your help!

Actually, I don't know exactly, but maybe the problem was, that I issued multiple `bleKeyboard.write()` commands without `delay()` between them.

However, I cannot tell if this was the reason or if it was something else...

closed 2019-11-14 by speendo