T-vK/ESP32-BLE-Keyboard #5: #include "sdkconfig.h" file or directory not found

ddunnam, 2019-11-21:
Disclaimer: This may very well be due to the fact I'm still fairly new to using Arduino controllers and the IDE in general.

I have the library is included in my project, but when I attempt to verify and upload I get the following error
`\Arduino\libraries\ESP32-BLE-Keyboard-0.1.0/BleConnectionStatus.h:3:10: fatal error: sdkconfig.h: No such file or directory`.

I am trying to upload to a Arduino MKR WiFi 1010 which uses an esp32 chip. I'm using IDE 1.8.8, and I have the esp32 board package version 1.0.4.

Any help is appreciated.

T-vK, 2019-11-21:
Did you follow all the installation instructions? Can you please try another ESP32 example? For instance, do the Wifi examples work for you?

ddunnam, 2019-12-11:
Sorry for the terribly delayed response.

I unzipped the ESP32-BLE-Keyboard library into my Arduino scripts folder. I actually found all the libraries the first step installed (just to be clear, this url [](https://github.com/espressif/arduino-esp32#installation-instructions)). I found the _sdkconfig.h_ in question and copied it into the folder where I extracted your library to.

That seemed to work, but it threw an error for the next header file. It looks like it can't find the esp libraries, but I've followed the steps to setup the esp libraries in the IDE, including installing _esp32 platform_ from boards manager.

When I compile an example in the esp library, it works just fine. In your library there's just the one _SendKeyStrokes_ example, and that's what I'm testing with.

T-vK, 2019-12-11:
That file doesn't belong into this library. 
sdkconfig.h is installed on your system if https://github.com/espressif/arduino-esp32/blob/master/README.md
has been set up correctly.

If it has been installed correctly, then it should be somewhere in the hardware directory under
hardware/espressif/esp32

maybe in
hardware/espressif/esp32/tools/sdk/include/config/sdkconfig.h

But I can't check at the moment.
The compiler will find that file automatically. Having a copy of it in your project directory won't help and will only do harm if anything.

Also, don't extract the library manually unless you know what you are doing. You only had to install it as I wrote in the instructions (Step 3), nothing more is required.

T-vK, 2019-12-12:
The file is in `/home/fedora/Arduino/hardware/espressif/esp32/tools/sdk/include/config/` on my Linux system. I assume you are on Windows, so for you instead of `/home/fedora/Arduino` is would be something different. Check in your IDE Preferences (File->Preferences -> (Sketchbook location)):
[image]

Please check if `sdkconfig.h` is in there.

ddunnam, 2019-12-13:
The esp library was installed into my `Users\Me\AppData` directory. So I tried copying it into my sketchbook; so I now have `C:\Users\Drew\Documents\Arduino\hardware\esp32\...`. I was still getting the same error. However, in the IDE I switched from having the Arduino MKR board selected to the _Wemos MINI D1 ESP32_ board which was specifically mentioned on the esp32 library in the board manager. This time I got these errors.

...\BleKeyboard.cpp:49:14: error: 'HIDOUTPUT' was not declared in this scope
   HIDOUTPUT(1),       0x02,          //   OUTPUT (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
...\BleKeyboard.cpp:52:14: error: 'HIDOUTPUT' was not declared in this scope
   HIDOUTPUT(1),       0x01,          //   OUTPUT (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
...\BleKeyboard.cpp:60:13: error: 'HIDINPUT' was not declared in this scope
   HIDINPUT(1),        0x00,          //   INPUT (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
...\BleKeyboard.cpp:88:13: error: 'HIDINPUT' was not declared in this scope
   HIDINPUT(1),        0x02,          //   INPUT (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)`

I haven't looked that yet, but it seems the MKR WiFi 1010 may be incompatible with the esp library although it has an esp32 on it. I'm not sure other that just selecting it in the boards menu how to get it to work.

T-vK, 2019-12-13:
I'm not sure which version I have installed, but I will try to upgrade and see if that breaks things.

Does it still compile if you remove the folder from your sketchbook again? (So that it's only in Users\Me\AppData)

I think you're probably right about the MKR WiFi 1010 not being compatible with normal ESP32 boards. The board probably has a normal Arduino which simply communicates with the ESP32 using a serial connection or something like that. So if you flash your code it will go onto the Arduino chip rather than the ESP32 chip.

You could probably flash the ESP32 chip directly, but then you would lose functionality like accessing the pins. 
If you would dig really deep, you could probably make it work, but it would require a lot of effort.

My advice is to buy a proper ESP32 dev board instead. They are only like $4 including shipping anyway.

ddunnam, 2019-12-13:
That’s what I was starting to assume, that the ESP32 was "hidden" behind the Arduino.  I feel like I tried removing it from the sketchbook to see if it still compiles, but I don’t remember how that turned out. I’ll update this post when I check on that.  From: T-vKSent: Friday, December 13, 2019 01:24To: T-vK/ESP32-BLE-KeyboardCc: ddunnam; AuthorSubject: Re: [T-vK/ESP32-BLE-Keyboard] #include "sdkconfig.h" file or directory not found (#5) I'm not sure which version I have installed, but I will try to upgrade and see if that breaks things.Does it still compile if you remove the folder from your sketchbook again? (So that it's only in Users\Me\AppData)I think you're probably right about the MKR WiFi 1010 not being compatible with normal ESP32 boards. The board probably has a normal Arduino which simply communicates with the ESP32 using a serial connection or something like that. So if you flash your code it will go onto the Arduino chip rather than the ESP32 chip.You could probably flash the ESP32 chip directly, but then you would lose functionality like accessing the pins.If you would dig really deep, you could probably make it work, but it would require a lot of effort.My advice is to buy a proper ESP32 dev board instead. They are only like $4 including shipping anyway.