T-vK/ESP32-BLE-Keyboard #7: bluetooth connecting trouble with Mac

lolololololololololololololololo, 2019-12-02:
so, first of all, I really enjoyed working with the library, it makes everything very clear end easy. but I have a problem when I try to connect to my MacBook Pro (Retina 13 inch, early 2015).
it just keeps on loading and then after a few seconds just stops. 
Somehow it got it to connect once ore twice but only those times.
But when I try to connect it to my iPhone (iPhone 7 ) it works like a charm.
I am using this board:
https://www.joy-it.net/de/products/SBC-NodeMCU-ESP32

her is the code I wrote: 
`#include <BleKeyboard.h>

BleKeyboard bleKeyboard("esp 32 Macro", "lolo", 100);
const int buttonPin0 = 2;
const int buttonPin1 = 4;
const int buttonPin2 = 18;
const int buttonPin3 = 19;
const int buttonPin4 = 21;
    

int buttonState0 = 0;  
int buttonState1 = 0;
int buttonState2 = 0;
int buttonState3 = 0;
int buttonState4 = 0;

int lastButtonState0 = LOW;
int lastButtonState1 = LOW;
int lastButtonState2 = LOW;
int lastButtonState3 = LOW;
int lastButtonState4 = LOW;

unsigned long lastDebounceTime0 = 0;  
unsigned long debounceDelay0 = 50;

unsigned long lastDebounceTime1 = 0;  
unsigned long debounceDelay1 = 50;

unsigned long lastDebounceTime2 = 0;  
unsigned long debounceDelay2 = 50;

unsigned long lastDebounceTime3 = 0;  
unsigned long debounceDelay3 = 50;

unsigned long lastDebounceTime4 = 0;  
unsigned long debounceDelay4 = 50;

void setup() {

 Serial.begin(115200);
 Serial.println("Starting BLE work!");
 bleKeyboard.begin();
  
  pinMode(buttonPin0, INPUT);
    pinMode(buttonPin1, INPUT);
      pinMode(buttonPin2, INPUT);
        pinMode(buttonPin3, INPUT);
          pinMode(buttonPin4, INPUT);
          
}

void loop() {
 
 
  int reading0 = digitalRead(buttonPin0);
  int reading1 = digitalRead(buttonPin1);
  int reading2 = digitalRead(buttonPin2);
  int reading3 = digitalRead(buttonPin3);
  int reading4 = digitalRead(buttonPin4);

  
if(bleKeyboard.isConnected()) {
  

if (reading0 != lastButtonState0) {
    
    lastDebounceTime0 = millis();
  }

  if (reading1 != lastButtonState1) {
    
    lastDebounceTime1 = millis();
  }

  if (reading2 != lastButtonState2) {
    
    lastDebounceTime2 = millis();
  }

  if (reading3 != lastButtonState3) {
    
    lastDebounceTime3 = millis();
  }
  
  if (reading4 != lastButtonState4) {
    
    lastDebounceTime4 = millis();
  }

  if ((millis() - lastDebounceTime0) > debounceDelay0) {
   
   
    if (reading0 != buttonState0) {
      buttonState0 = reading0;

     
      if (buttonState0 == HIGH) {
        bleKeyboard.print("2 ");
        
        Serial.println("0");
      }
    }
  }

  if ((millis() - lastDebounceTime1) > debounceDelay1) {
   
   
    if (reading1 != buttonState1) {
      buttonState1 = reading1;

     
      if (buttonState1 == HIGH) {
        bleKeyboard.print("4 ");
        Serial.println("1");
      }
    }
  }

  if ((millis() - lastDebounceTime2) > debounceDelay2) {
   
   
    if (reading2 != buttonState2) {
      buttonState2 = reading2;

     
      if (buttonState2 == HIGH) {
        bleKeyboard.write(KEY_LEFT_GUI);
        bleKeyboard.print("z");
        Serial.println("2");
      }
    }
  }

  if ((millis() - lastDebounceTime3) > debounceDelay3) {
   
   
    if (reading3 != buttonState3) {
      buttonState3 = reading3;

     
      if (buttonState3 == HIGH) {
        bleKeyboard.print("19 ");
        Serial.println("3");
      }
    }
  }

  if ((millis() - lastDebounceTime4) > debounceDelay4) {
   
   
    if (reading4 != buttonState4) {
      buttonState4 = reading4;

     
      if (buttonState4 == HIGH) {
        bleKeyboard.print("21 ");
        Serial.println("4");
      }
    }
  }

 
   lastButtonState0 = reading0;
    lastButtonState1 = reading1;
     lastButtonState2 = reading2;
      lastButtonState3 = reading3;
       lastButtonState4 = reading4;
 }
}`
maby someone can help me. I would realy appreciate it 
best reguards Lorenz

T-vK, 2019-12-02:
Does your Mac support Bluetooth LE? https://help.kano.me/hc/en-us/articles/360002246660-Checking-Bluetooth-Low-Energy-Support-on-Your-Device

lolololololololololololololololo, 2019-12-02:
thanks for your quick reply. 
yes it does support Bluetooth LE

lolololololololololololololololo, 2019-12-02:
do you have any other idea what the problem could be?

T-vK, 2019-12-03:
Unfortunately I don't know. Could be a bug in the underlying BLE library that I'm using or maybe a bug in OS X. It's really hard to say for me, especially because I don't own any Apple devices.

Maybe it's an [authentication issue](https://github.com/nkolban/esp32-snippets/issues/230#issuecomment-554595413).

lolololololololololololololololo, 2019-12-03:
ooh, I think I maybe got it.
I'm using an OS X Beta version. This is probably it.
tank you very much for your help though. :D

lolololololololololololololololo, 2019-12-03:
Il try it on another Mac or PC, then I can be for certain 👍

lolololololololololololololololo, 2019-12-04:
I got it. My mac does not support Bluetooth 4.2, only 4.1 and the esp uses 4.2.
Do you know if it is possible in any way to use the esp 32 with Bluetooth 4.0?

lolololololololololololololololo, 2019-12-04:
maby if I use an esp that uses Bluetooth 4.0 ? do you know if the library works with Bluetooth 4.0?

T-vK, 2019-12-04:
The version itself shouldn't matter. It all comes down to Bluetooth Low Energy support. If your device uses a Bluetooth 4.x implementation that doesn't support BLE, then it won't work. Otherwise it should work fine, even if it was Bluetooth 5.x.

lolololololololololololololololo, 2019-12-05:
nooo :( I thought I head the problem.
then I am back to troubleshooting because my mac supports BLE.
[image]
I know this is german but it essentially says BLE is supported.
I am going to try it with another device and then if it still won't work maby I am going to buy another esp.

T-vK, 2019-12-05:
Since you're able to connect your ESP with your iPhone, I highly doubt there is anything wrong with your ESP. It's probably a software issue. Maybe it's because you're using the OS X Beta. Or maybe it's incorrectly reporting that BLE is supported. Perhaps it would work with a USB Bluetooth dongle, but I don't know if that would help or which ones work on OS X in the first place.
On Linux, this one is working great for me: https://www.aliexpress.com/item/32696756077.html (even though it says it doesn't support Linux and OS X)

lolololololololololololololololo, 2019-12-06:
ok, that sounds plausible. I'm going to try it with a dongle. 
thank you very much for your support  :)

lolololololololololololololololo, 2019-12-15:
hello, it is me again :) 
so I have done some testing. 
I have tested the esp wit two different macs which are both much newer than mine, but I had no success. 
so I think that the problem must be with the library ore with a mac authentification problem.
do you know of anything that could be done to fix this problem? I would really appreciate it.

leachiM2k, 2019-12-18:
Do you use OS X Catalina? Before the upgrade it worked, but after the upgrade I can’t even connect.

lolololololololololololololololo, 2019-12-21:
omg, I just fixed it. 
this did it for me:  
[image]
thank you so so mutch for your help and pations…

aerialist, 2020-05-31:
Excellent discussion!

I faced reconnection problem on my MacBook Pro with macOS Catalina. Suggested authentication issue seems to be the cause and ossvr's workaround worked for me.

I raised PR.
#28

chienr, 2020-08-31:
Although this resolves the connection issue for me as well on Catalina (10.5.5), it seems to cause some noticeable lag in typing response and also lag in another bluetooth connected mouse. Do you experience the same?