T-vK/ESP32-BLE-Keyboard #10: Works on MAC OSX Mojave 10.14.4 out of the box.

wyojustin, 2020-02-02:
Tries to delete itunes data LOL

T-vK, 2020-02-03:
@wyojustin What do you mean?

wyojustin, 2020-02-03:
Sorry for the terse note.  I see in your wiki that mac is untested.  I can confirm pairing with a mac and android now.   This issue is just to let you know this information.

wyojustin, 2020-02-03:
Oh yeah... the LOL.  The script hits play (on mac this pulls up itunes).  Then it tries to CTRL-ALT-DEL which tries to delete the default file in itunes (LOL I don't use itunes).

closed 2020-02-03 by wyojustin

T-vK, 2020-02-04:
Oh, I see. Thanks for reporting that. That's definitely not good. Does it just open a prompt or does it actually delete the file?

wyojustin, 2020-02-04:
Opens a prompt.

T-vK, 2022-01-06:
This issue has been fixed by ScaredyCat in #134 and #135. It's on the master branch and will make it into the next beta release.