T-vK/ESP32-BLE-Keyboard #11: Sending Fkeys over F12

tiking, 2020-02-04:
I tried to send the F13 key (KEY_F13 code) but it did not work (not received on the Windows 10 side). But F12 (KEY_F12 code) and lower is working correctly.

tiking, 2020-02-04:
You can close the issue, I solved it the same way it was corrected in the standard Keyboard.cpp file.

I had to modify those 2 lines to increase the maximum from 0x65 to 0x73 in the BleKeyboard.cpp file :  

LOGICAL_MAXIMUM(1), 0x65,          //   LOGICAL_MAXIMUM(0x65) ; 101 keys
USAGE_MAXIMUM(1),   0x65,          //   USAGE_MAXIMUM (0x65)

I don't know if it could be modified in your source files directly.

Regards,

RyanBode, 2020-04-12:
I have modified the two lines you list above and still can't seem to get this working? Is there anything else I need to edit?

opigir, 2020-07-02:
> I have modified the two lines you list above and still can't seem to get this working? Is there anything else I need to edit?

Hey Ryan I was just tearing my hair out over this exact issue. Once you have modified the 
LOGICAL_MAXIMUM(1), 0x65, // LOGICAL_MAXIMUM(0x65) ; 101 keys
USAGE_MAXIMUM(1), 0x65, // USAGE_MAXIMUM (0x65)

You must Remove the ble device from your computer settings or else it will not update the keyboard mapping and f13 - f24 will not show as it will keep the old profile it initial had. 

Once you re-add the ble device it should work as expected. )This took me 4 hours to work out!

ignis32, 2021-12-02:
Workaround worked for me, after fixing both parameters and re-paring device.
May we have this fix in the original library code?

T-vK, 2022-02-15:
I would accept a Pull Request for this, if it gets tested on iOS, Android, Windows, Linux and Mac OS.