T-vK/ESP32-BLE-Keyboard #12: Media Keys on Android do not work after reconnect BLE

mini1053, 2020-02-04:
I send media key commands and it works fine on my Android device.
But if connection is lost (to far away) or ESP32 is restarted the BLE connection is reestablished but my Android device will not get the media key information anymore/or does not react to it. I can send text with print or characters with write and this works fine.
With mit IOS device everything works fine even after lot connection and reconnect.
If I do a new pairing between Android device and ESP it works again fine.
Tested now on two different Android devices. 
Do you have any clue why media key commands do not work after reconnect?

T-vK, 2020-02-05:
That's very odd. Can you tell me which devices you tested this on and tell me their Android/iOS version?
Did this happen with one of the provided examples? (If not, I'd like to check your code.)

For now, my guess is that Bluetooth on Android is just buggy. I've had issues with Bluetooth headsets on several Android devices that often required re-pairing after a disconnect.

mini1053, 2020-02-05:
Hello

I have following setup:
  ESP32 Board: Adafruit HUZZAH32 feather with WROOM-32
   
  5 buttons connected to GPIO 14, 32, 15, 33, 27
  using internal pull-ups
  GPIO activated if button connected to GND 

My device is used for for the electronic rally road book project for
scrolling the road-book and correcting the oddometer for motorcycle
rallying. Compatible to MDR road book, or Rally Roadbook App.
 
  Android or IOS device uses to display road book etc. 
   

Carpe Iter Pad
  Android 5.1
  works after pairing inclusive media Buttons.
  reconnecting (restaring ESP32)works after 2-5 seconds, key commands are
transfered but no media keys.
  Bluetooth off/on on Android device same effect
  after new pairing it works fine again inclusive media Buttons.
   
  Samsung Galaxy Tab A SM-T380
  Android 9
  works after pairing inclusive media Buttons.
  reconnecting (restaring ESP32) works after 2-3 seconds, key commands are
transfered but no media keys.
  Bluetooth off/on on Android device works fine inclusive media Buttons, no
repairing needed
   
   
  Redmi Note 6 Pro
  Android 9
  works after pairing inclusive media Buttons.
  reconecting (restaring ESP32) but imedeately after reconnecting it
disables Bluetooth on device (strange ???) after enabeling Bluetooth again
it works inclusive media Buttons.
  Why can a reconnection of a Bluetooth connection disable the Bluetooth
function on phone?

BR
Gerald

Links:
------
[1]  
https://github.com/T-vK/ESP32-BLE-Keyboard/issues/12?email_source=notifications&email_token=AOO35LQLYXHCQKUEJGNKP2TRBJ5ODA5CNFSM4KP4RWO2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEK2UOFQ#issuecomment-582305558
[2]  
https://github.com/

mini1053, 2020-02-05:
Hello,

yes, same problem with your example code 

BR
Gerald

Links:
------
[1]  
https://github.com/T-vK/ESP32-BLE-Keyboard/issues/12?email_source=notifications&email_token=AOO35LQLYXHCQKUEJGNKP2TRBJ5ODA5CNFSM4KP4RWO2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEK2UOFQ#issuecomment-582305558
[2]  
https://github.com/

mini1053, 2020-02-07:
Hello, I have found following example and there the media keys work after
reconnecting on Android.

https://gist.github.com/ematt/29983b14381444ce1081177c65e22010[1]

BR
Gerald

Links:
------
[1]  
https://www.youtube.com/redirect?stzid=UgyV7a8ZDn_DGR6d69h4AaABAg.93Z8O_kx69l94hmGRikzG6&q=https%3A%2F%2Fgist.github.com%2Fematt%2F29983b14381444ce1081177c65e22010&event=comments&redir_token=Kpe8Ico92fDwGRuS2fJF-18AKNV8MTU4MTE1MjA2MEAxNTgxMDY1NjYw
[2]  
https://github.com/T-vK/ESP32-BLE-Keyboard/issues/12?email_source=notifications&email_token=AOO35LQLYXHCQKUEJGNKP2TRBJ5ODA5CNFSM4KP4RWO2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEK2UOFQ#issuecomment-582305558
[3]  
https://github.com/

xillus1, 2020-02-15:
The problem is on class BleConnectionStatus, in the callbacks methods. You must set notification true or false also for the inputMediaKeys.

T-vK, 2020-02-27:
Thanks to @mwinters-stuff for fixing this in #16 . This is now fixed in release 0.2.1.

closed 2020-02-27 by T-vK

KrumpsProjects, 2025-02-14:
I was having the same issue with the 0.3.0 release. But then jumped back to the 0.2.1 and it appears to be reconnecting properly. So something got broke in between there. I'm using a Samsung Galaxy A32 5G and ESP32 DevkitV1 ESP-WROOM-32 with 30 pins.