T-vK/ESP32-BLE-Keyboard #14: Work with ESP32-BLE-Mouse?

blackketter, 2020-02-13:
I can't seem to get the ESP32-BLE-Keyboard and ESP32-BLE-Mouse to work together in the same sketch.

I've made some modifications to BleConnectionStatus to have unique BleMouseConnectionStatus and BleKeyboardConnectionStatus files and classes, but it seems that only one of the endpoints is detected on the host.  Any tips on how to make them work together?  Happy to submit a patch once I've got it working...

T-vK, 2020-02-14:
Good question. I haven't tried that yet, but it would indeed be nice if those libraries worked together. 
I'm not sure if the underlying Bluetooth library even allows for multiple devices at the same time, but what you could do is merge the two `_hidReportDescriptor`s. Then you'd have one Bluetooth device with keyboard and mouse capabilities. But you'd also have to merge the mouse and keyboard methods and adjust things for the new hid descriptor...

blackketter, 2020-02-14:
It seems that's the case. I'm working on trying to merge them now.  What would you think of a combo library with classes for different combinations to avoid code duplication?

ankitpatle, 2020-02-15:
@blackketter Any luck at testing after merge??

blackketter, 2020-02-16:
Sorry, I’m out of town for a couple of days.  When I left it was crashing, I’ll continue to poke at it when I get back

T-vK, 2020-02-16:
Well, the two libraries have been written with the intention of creating drop-in replacements for the original usb "keyboard" and "mouse" libraries. 
I also feel like they should be maintained in separate repositories. It might however be a good idea to create a common submodule for all that redundant BLE stuff.

In case it is only possible to get these to work together using a common hid descriptor, we should, in my opinion,  create a separate repository for that because there may be edge cases where having one device show up as multiple devices causes a problem.
Ideally we'd get the underlying BLE library patched to support multiple devices at the same time, if it doesn't already.

ankitpatle, 2020-02-17:
So, I was looking in the bluetooth descriptor IDs and from [this document](https://www.google.com/url?sa=t&source=web&rct=j&url=https://cdn.sparkfun.com/datasheets/Wireless/Bluetooth/RN-HID-User-Guide-v1.0r.pdf&ved=2ahUKEwiq9IO0sdjnAhWhguYKHRDJBLAQFjAAegQIAhAB&usg=AOvVaw1eZUgVHJB6jM-GGELRNXL9), got these,

Descriptor type:
0000 = Keyboard
0001 = Game Pad
0010 = Mouse
0011 = COMBO
0100 = JOYSTICK
0101 = DIGITIZER
0110 = SENSOR
0111 = USE CFG
1XXX = Reserved

Some keyboard+mouse devices use the Combo Descriptor. So i think maybe we can use that. I will check that once i have some free time from my exams.

blackketter, 2020-02-22:
So, as an experiment, I've forked this library and made a combo mouse and keyboard library here: https://github.com/blackketter/ESP32-BLE-Combo

Not sure if this is the right approach long-term but it seems to be working well on my Mac and iPhone.  Would love feedback.  This could be extended to add joystick and other descriptor types.

I didn't use the COMBO descriptor listed above, just concatenated the two descriptors and created a separate mouse class that uses the keyboard class' bluetooth interface.

I did notice that the Arduino USB mouse and keyboard libraries have globally defined Keyboard and Mouse objects, I'm not sure that's the right approach here, but I thought I'd note it.

lemmingDev, 2020-03-30:
Hey @blackketter and @T-vK 

Am hacking around both the mouse library and the combo library to turn the mouse into an absolute mouse for making a Bluetooth lightgun.

Converted the ESP32-BLE-Mouse library no problem, but would now like to convert the ESP32-BLE-Combo library too to have extra functionality, but am getting stuck soooo close to getting it working perfectly.

The issue I'm having is that everything works perfectly when it's paired the first time after programming, however when it reconnects after being off, or after reprogramming, only the keyboard functionality works.

If I remove the BT device in Windows BT settings and the re-add it, it works fine again, until I turn BT off and off again and it reconnects, but then only the keyboard part of it works. 
Remove the device, re-add, and then it works fine again.

My question - would one of you be willing to look at the combo code for me to potentially fix?
[ESP32-BLE-Combo.zip](https://github.com/T-vK/ESP32-BLE-Keyboard/files/4402502/ESP32-BLE-Combo.zip)

[ESP32-BLE-MouseAbs.zip](https://github.com/T-vK/ESP32-BLE-Keyboard/files/4402505/ESP32-BLE-MouseAbs.zip)

lemmingDev, 2020-03-31:
Oh - just realised that the @blackketter ESP32-BLE-Combo library has the same issue

Can anyone confirm the issue? 

If I turn the system Bluetooth off, and then on again, it works, but if I turn the ESP32 off and on again, only the keyboard part of it works, until I remove the BT combo device pairing from Windows,  and re-add it...

chibisuke, 2020-05-28:
@lemmingDev I can confirm this issue, but this is out of scope on this repository. However there is an easy fix to it. (https://github.com/blackketter/ESP32-BLE-Combo/pull/2)

lemmingDev, 2020-09-10:
> @lemmingDev I can confirm this issue, but this is out of scope on this repository. However there is an easy fix to it. ([blackketter#2](https://github.com/blackketter/ESP32-BLE-Combo/pull/2))

Forgot to say --> thanks heaps