T-vK/ESP32-BLE-Keyboard #15: Smooth keyboard injections

porchus, 2020-02-20:
Hi, I am trying to inject keystrokes in a continuous manner. The problem is that I either stuff the buffer and when I release the key it keeps on sending, or it I sende too few keystrokes. I basically use trial and error, but is there a technical explanation ot the max times, delays, etc.?

T-vK, 2020-02-21:
Can you provide a code example? I don't fully understand your issue.

porchus, 2020-02-21:
Basically I want to generate movement by emulating the arrow key.

      bleKeyboard.write(KEY_UP_ARROW);
      delay(delayStep);

That's working fine, the point is that I want to optimize my variable delayStep. I want to inject these keystrokes continuously (emulating keeping my finger pressed on the key). I tried diferent values but cannot get a smooth movement on my program, whereas I do get one when I hold down the key in the actual keyboard.
Thats why I want to know how to calculate the minimum delay time per keystroke.

T-vK, 2020-02-21:
The `.write(...)` method sends a press and release event. You only want to send a press event. 
Use the `.press(...)` method instead and once you want to release the key again, use the ` .release(...)` method.

porchus, 2020-02-21:
Thanks!!!! Problem solved. Now it runs really smooth. I'd like to thank you for the following:
1) A prompt response.
2) Helping me solve the problem.
3) A DAMN good solucion.

T-vK, 2020-02-22:
No problem.

closed 2020-02-22 by T-vK