T-vK/ESP32-BLE-Keyboard #20: Sketch has no space for Wifi Libriary

quietbiker, 2020-05-01:
Firstly, this is a great Library - thanks!

This may be down to a cheap ESP32 but with just the Library and an even smaller program (just sending Vol-Up) it's not leaving much spare storage.

> Sketch uses 1005159 bytes (76%) of program storage space. Maximum is 1310720 bytes.

I was hoping to include the Wifi stuff, but there isn't enough space.

My own project is simply to trigger a photo (Vol-up appear to work on Android and iPhone).

T-vK, 2020-05-01:
I haven't verified if it is true, but if it is, there is not much I can do about it. Using the official ESP32 sdk instead of the Arduino IDE would probably reduce the size. But I couldn't help you with that.

quietbiker, 2020-05-01:
Ok, thanks for the reply.  I do appreciate the project.

I'm remotely triggering a phone camera whilst I index a turntable (driven by the ESP32) to see if I can get acceptable 3D scans using Meshroom.
Wifi isn't essential as the ESP is really just triggering the shutter over BT and driving a the Stepper IC.

Thanks again!

closed 2020-05-01 by quietbiker

uolot, 2020-05-01:
@quietbiker Read about partition tables for ESP32, disabling OTA gives you some extra space and is supported in ESP-IDF, Platformio and Arduino IDE (search for specific environment for more details).

quietbiker, 2020-05-01:
Awesome!  Yes, all running with Wifi.
I've never spotted that before.

Thanks for your help @uolot