T-vK/ESP32-BLE-Keyboard #21: Power optimization

itonik, 2020-05-04:
Thanks for great library :)

Is there any known method to make power consumption lower?
WROOM devkit, keyboard idle -> around 60-65mA, same when sending keypresses.
As far as I understand how BLE works, it shoud not send data when no data has to be transfered.

I cannot use light sleep because I shoud turn off bluetooth and bluedroid prior to going to sleep afaik.

T-vK, 2020-05-04:
There is no method I know of. The ESP32 is relatively power hungry compared to other BLE devices and Espessif doesn't seem to be interested in optimizing BLE for power efficiency.

itonik, 2020-05-04:
After some digging, I found two facts:
1. Efficient enough ble is only possible with external crystal
2. It will not work in arduino IDE (as far as i understand, because of how loop() implemented).
Closing issue, hopefully this information will be helpful for someone :)

closed 2020-05-04 by itonik

pepsi-zero, 2020-07-18:
You can reduce the Power consumption, by reducing the frequency the ESP32 is running.
This can be done without an external crystal. 
https://www.savjee.be/2019/12/esp32-tips-to-increase-battery-life/#tip-4-reduce-the-clock-speed

I’ve run an M5atom via 5volt with just the keyboard code. And on 5V it was above the 50mA but at 160 i was onder the 50mA. And 50mA or les was the cutoff current for my 3.7 to 5V step up converter. 
I will check a esp32 that is not on a development board with an mpc1700. Form 3.7V to 3.3V.