T-vK/ESP32-BLE-Keyboard #23: F1/F2 Brightness for iOS

balsman225, 2020-05-06:
I am trying to use an ESP32 to control a wall mounted iPad screen brightness (plan on pairing this library with MQTT). 

On a normal Apple Bluetooth keyboard you can used F1 and F2 to adjust the brightness. 
I have looked over [Keyboard Modifier](https://www.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardmodifiers) 

I tried several combination without success
bleKeyboard.press(KEY_F1)
bleKeyboard.write(KEY_F1) 
bleKeyboard.press(194) 
bleKeyboard.write(194) 
bleKeyboard.write(0xC2) 

When testing with the "SendKeyStrokes" sketch I am able to get all other tests to work. 
KEY_MEDIA_VOLUME_UP and KEY_MEDIA_VOLUME_DOWN work fine as well. 

I'm curious if I need to create a custom "KEY_MEDIA" for brightness since iOS does not seem to be responding to F1/F2 keys 

Thank you for the assistance
Also - Thank you for the great library

T-vK, 2020-05-06:
Can you check if your keys are recognized in any way? https://www.keyboardtester.com/tester.html

balsman225, 2020-05-07:
Thanks for the pointer - I looked over #3 and tested something out - 
I changed the Volume increase and decrease to a custom brightness button.

I found the needed Hex for the Brightness Adjustment [Link Here](https://github.com/NicoHood/HID/blob/d4938ddcff7970bc1d32a040a08afeac4915e4a9/src/HID-APIs/ConsumerAPI.h#L154)

For Brightness Up - 
https://github.com/T-vK/ESP32-BLE-Keyboard/blob/63e08b72f2a2c5ffae00950cb17450e4986ee768/BleKeyboard.cpp#L77
with
```
    USAGE(1),           0x006F,        //   USAGE (Brightness Up)       ; bit 5: 32
```

For Brightness Down - 
https://github.com/T-vK/ESP32-BLE-Keyboard/blob/63e08b72f2a2c5ffae00950cb17450e4986ee768/BleKeyboard.cpp#L78
with
```
  USAGE(1),           0x0070,        //   USAGE (Brightness Down)     ; bit 6: 64
```

For Brightness Up - 
https://github.com/T-vK/ESP32-BLE-Keyboard/blob/63e08b72f2a2c5ffae00950cb17450e4986ee768/BleKeyboard.h#L68
with 
```
const MediaKeyReport KEY_MEDIA_BRIGHTNESS_UP = {32, 0};
```

For Brightness Down - 
https://github.com/T-vK/ESP32-BLE-Keyboard/blob/63e08b72f2a2c5ffae00950cb17450e4986ee768/BleKeyboard.h#L69
with 
```
const MediaKeyReport KEY_MEDIA_BRIGHTNESS_DOWN = {64, 0};
```

Once I added 
bleKeyboard.write(KEY_MEDIA_BRIGHTNESS_UP)
and 
bleKeyboard.write(KEY_MEDIA_BRIGHTNESS_DOWN)
to my test sketch It seems to be working great now. 

This is all new to me so I don't know how to help added this to as a new media key in the library instead of stealing the volume entries but I would be happy to test for you if needed.

T-vK, 2020-05-11:
I can't quite remember, but I think there was a reason why I only implemented a few media keys. Maybe it was a limitation in the hid descriptor or some kind of compatibility issue with the official Arduino USB keyboard library. If it was easy, I would of course have added way more, if not all media keys.
But thanks for sharing. This is probably gonna be useful to someone in the future.

theguaz, 2022-05-31:
@balsman225 @T-vK  Hello !, first of all nice work, I've found your solution while browsing for brightness control using BLe devices, I followed the changes u made in both files to basically swap the volume control with brightness control but so far when I swaped the values I still get Volume changes in iOS and OSX as well instead of brighness.

In CPP file:
`//const MediaKeyReport KEY_MEDIA_VOLUME_UP = {32, 0};
//const MediaKeyReport KEY_MEDIA_VOLUME_DOWN = {64, 0};

const MediaKeyReport KEY_MEDIA_BRIGHTNESS_UP = {32, 0};
const MediaKeyReport KEY_MEDIA_BRIGHTNESS_DOWN = {64, 0};
`
In H file:
`  USAGE(1),           0xCD,          //   USAGE (Play/Pause)          ; bit 3: 8
  USAGE(1),           0xE2,          //   USAGE (Mute)                ; bit 4: 16
  USAGE(1),           0x006F,        	//   USAGE (Bright Increment)    ; bit 5: 32
  USAGE(1),           0x0070,        	//   USAGE (Bright Decrement)    ; bit 6: 64
  USAGE(2),           0x23, 0x02,    //   Usage (WWW Home)            ; bit 7: 128`
  
  
  The hexadecimal needs to be edited somewhere else in order to trigger brightness?
  Thank you in advance :)

TanjuDuygu, 2023-02-15:
Hi balsman. 

You are on the right way

USAGE(1),           0x006F,        	//   USAGE (Bright Increment)    ; bit 5: 32
USAGE(1),           0x0070,        	//   USAGE (Bright Decrement)    ; bit 6: 64

 You need just need unpair and agin pair your bluetooth device pair your bluetooth device on your iphone. I treid %100 working birightness on my iphone ind ipad. If you din't refresh pair still stick working just volume up and down. 
Good luck

and7ey, 2023-12-05:
@balsman225 sorry for off top question - is it possible to switch off the iOS screen with ESP32 keyboard? or to lock the iPad?