T-vK/ESP32-BLE-Keyboard #25: Trouble connecting with Android Samsung Galaxy Note 10

vcchm, 2020-05-22:
Hi !

Thanks a lot for this library, really helpful. I used it to make a dongle that send very strong passwords (32 characters, fully random patterns).

I used your library just adding a pincode security to avoid someone connecting to the dongle and stealing the password

  BLESecurity *pSecurity = new BLESecurity();
  pSecurity->setStaticPIN(bleKeyboardInstance->hid_pincode);
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);  

it works like a charm on my windows computer, but not on my android phone (Samsung Galaxy Note  SM-N950F, Android 9). It was working with android previous version, i think it is probable related to a security update.

When I do the pairing i get the following message : "Pairing impossible." It works sometimes if I delete the pincode setting line.
 Any idea regarding what's wrong ? Also, can you tell me where I can find the documentation regarding authentificationmode , setCapability and setInitEncryptionKey ?

vcchm, 2020-05-22:
I tried with other devices:

Motorola : can't connect. Message = "invalid code"

T-vK, 2020-05-25:
Sorry, I haven't touched any of the alternative authentication methods yet and don't really have time looking into it any time soon. 
I don't think there is documentation for that. You probably have to dig throught the source code.  https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLESecurity.cpp#L20
My understanding is that the BLE libraries that we use with the Arduino IDE are basically wrappers around the official ESP32 SDK. E.g. look for esp_ble_gap_set_security_param in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gap_ble.html

vcchm, 2020-05-27:
Thanks. This is beyond my knowledge too :-(