T-vK/ESP32-BLE-Keyboard #27: Additional media keys Creation

DhruvGupta5, 2020-05-30:
Hi,
tested the library with esp32 and ios 12, 13 ,10 everything works as told, extremely thankful for putting such a wonderful product together.
i wanted to have some information as to how to extend "send media keys" "KEY_MEDIA" like maybe ability to add screenshot, launch voice recorder (primarily on ios) etc.
if you can point me to right direction as to where to begin i maybe able to extend it further, any help would be great

T-vK, 2020-05-30:
Take a look at #3 #11 and #23

DhruvGupta5, 2020-05-30:
> Take a look at #3 #11 and #23

Thanks a lot, its been great help, 
feel foolish not going through existing question:(

T-vK, 2020-05-30:
No problem, it happens. I will close this issue as it's basically a duplicate of #3.

closed 2020-05-30 by T-vK