T-vK/ESP32-BLE-Keyboard #31: IOS13 Microphone mute button

dotanitis, 2020-07-21:
Hello to all,

First I would like to thank the author of this great project.
I would like to send a command to my IOS13 device to mute the microphone during conversation, is the command is available?

T-vK, 2020-07-21:
I don't know, did you try KEY_MEDIA_MUTE?

dotanitis, 2020-07-21:
Yep. It just mute the speakers. 
(I've also tried to use the mouse BLE but unable to send any commands. The motivation was to send click command in the icon x,y)

T-vK, 2020-07-22:
Right, my bad. I'm not aware of any media keys that would mute the microphone. And since I don't own any Apple devices I also don't know if there are keyboard shortcuts to mute the microphone. 
If you can get ESP32-BLE-Mouse to work I'd assume it should be possible with enough tinkering to get the mouse to click the icon. It won't be as easy as saying "click at position x,y" though. Coorodinates are always relative to the current mouse position and there is no way to query the current mouse position. Also, coordinates are percentual and there is no way to query the screen resolution.