T-vK/ESP32-BLE-Keyboard #33: SNES Control

aieseres, 2020-07-28:
Sorry for my complete ignorance, I'm new to this world of Arduino and ESP32.
I thought this work was great and I'm trying to adapt it for a SNES control. For each button on the control I am assigning a key, for button "A" on the control for example, assign the key "x" and sending this data via BLE to the computer with ZSNS. My problem seems to be in the timing, because when pressing a button on the control there is no instant response on the emulator, I have to press several times to enter the command. Was there a way to sync the keyboard input?

Below my code.

#include <BleKeyboard.h>

BleKeyboard bleKeyboard;

//define where your pins are
const int dataPin = 21;  // RED
const int latchPin = 19; // YELLOW
const int clockPin = 18; // BLUE
const int intervalo = 250; // atraso entre leituras

long buttonData, lastButtonData = 72;  //01001000

// On NES Controller there are only 8 buttons, also B, Y swap to A and B
const int n_buttons = 12;
char* buttons[] = {"B", "Y", "SELECT", "START",
                   "UP", "DOWN", "LEFT", "RIGHT",
                   "A", "X", "L_BUTTON", "R_BUTTON"};

//Funções que serão chamadas por um array de pinteiros
void Button_B() {/*Serial.println("Apertou o B bruxo!!!!");*/              bleKeyboard.print("z");              delay(intervalo);}
void Button_Y(){/*Serial.println("Apertou o Y bruxo!!!!");*/               bleKeyboard.print("a");              delay(intervalo);}
void Button_SELECT(){/*Serial.println("Apertou o SELECT bruxo!!!!");*/     bleKeyboard.write(KEY_RIGHT_SHIFT);  delay(intervalo);}
void Button_START(){/*Serial.println("Apertou o START bruxo!!!!");*/       bleKeyboard.write(KEY_RETURN);       delay(intervalo);}
void Button_UP(){/*Serial.println("Apertou o UP bruxo!!!!");*/             bleKeyboard.write(KEY_UP_ARROW);     delay(intervalo);}
void Button_DOWN(){/*Serial.println("Apertou o DOWN bruxo!!!!");*/         bleKeyboard.write(KEY_DOWN_ARROW);   delay(intervalo);}
void Button_LEFT(){/*Serial.println("Apertou o LEFT bruxo!!!!");*/         bleKeyboard.write(KEY_LEFT_ARROW);   delay(intervalo);}
void Button_RIGHT(){/*Serial.println("Apertou o RIGHT bruxo!!!!");*/       bleKeyboard.write(KEY_RIGHT_ARROW);  delay(intervalo);}
void Button_A(){/*Serial.println("Apertou o A bruxo!!!!");*/               bleKeyboard.print("x");              delay(intervalo);}
void Button_X(){/*Serial.println("Apertou o X bruxo!!!!");*/               bleKeyboard.print("s");              delay(intervalo);}
void Button_L_BUTTON(){/*Serial.println("Apertou o L_BUTTON bruxo!!!!");*/ bleKeyboard.print("d");              delay(intervalo);}
void Button_R_BUTTON(){/*Serial.println("Apertou o R_BUTTON bruxo!!!!");*/ bleKeyboard.print("c");              delay(intervalo);}

//Array com os ponteiro que chamará as funções
void (*ButtonPointer[])(void) = {Button_B, Button_Y, Button_SELECT,  Button_START,
                                  Button_UP, Button_DOWN, Button_LEFT, Button_RIGHT,
                                  Button_A, Button_X, Button_L_BUTTON, Button_R_BUTTON};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();

  //define pin modes
  pinMode(latchPin, OUTPUT);
  pinMode(clockPin, OUTPUT); 
  pinMode(dataPin, INPUT);

//Cria a Task para ler os botões do controle SNES ****************************************************
    xTaskCreate(
                    taskControleSNES,          /* Task function. */
                    "TaskControleSNES",        /* String with name of task. */
                    10000,            /* Stack size in bytes. */
                    NULL,             /* Parameter passed as input of the task */
                    1,                /* Priority of the task. */
                    NULL);            /* Task handle. */
//****************************************************************************************************

  
}

void loop() {
   if(bleKeyboard.isConnected()) {

  //if (buttonData != lastButtonData){ 
    lastButtonData = buttonData;
     //Print Raw Value of Registers
    //Serial.println(buttonData, BIN);
  
    // Print String in Button Name Array
    for (int n = 0; n < n_buttons; n++)
    {
      if (buttonData & (1 << n) ){
        //Serial.print(buttons[n]);
        ButtonPointer[n]();
        //Serial.print(", ");
      }
    }
    //Serial.println("\n-------------------");
  //}

    //delay(10);
      
    //bleKeyboard.releaseAll();
   }
}

void taskControleSNES( void * parameter )
{

  while(1){
  
    // Latch Signal
  digitalWrite(latchPin, HIGH);
  delayMicroseconds(12);
  digitalWrite(latchPin, LOW);
  
  int temp = 0;
  buttonData = 0;
  for (int i = 0; i < n_buttons; i++)
  {
    digitalWrite(clockPin, LOW);
    delayMicroseconds(0.2);
    temp = digitalRead(dataPin);
    // 0 Indicates button depressed for SNES controller
    if (!temp) {
      //temp = size_buttonData - i;
      buttonData = buttonData | ((long)1 << i);
    }
    digitalWrite(clockPin, HIGH);
  }

  //if (buttonData != lastButtonData){ 
    //lastButtonData = buttonData;
    // Print Raw Value of Registers
    //Serial.println(buttonData, BIN);
  
    // Print String in Button Name Array
   /* for (int n = 0; n < n_buttons; n++)
    {
      if (buttonData & (1 << n) ){
        Serial.print(buttons[n]);
        Serial.print(", ");
      }
    }
    Serial.println("\n-------------------"); */
  //}
  delay(10);
  }
          
  vTaskDelete( NULL );
 
}

T-vK, 2020-07-28:
I don't think you're having an issue with the library itself. Quickly skimming through your (horribly formatted) code, it seems likely that your use of the delay function causes these issues. The delay function completely freezes code execution preventing new button presses to be registered until the delay is over.
You might wanna google how to avoid the usagle of delay().

aieseres, 2020-07-29:
Thanks for your patience with me, I'm very newbie. Yes, maybe the delay is the problem. But how do I know if the first key is completely sent by BLE and can I sent new one?

T-vK, 2020-07-29:
That shouldn't really matter, right? I mean on a physical controller you can press a button while holding down another one, or you can press two buttons at the same time.

aieseres, 2020-08-02:
The problem seems to be with holding a button. For example, to move Mario forward I need to hold the right key on a regular keyboard. But if I enter the command 'bleKeyboard.write (KEY_RIGHT_ARROW);' in the loop, it is as if I pressed the right key many times, not like holding the key. This happens for all other keys.

T-vK, 2020-08-02:
Why would you do bleKeyboard.write though? When you detect a button press you should do bleKeyboard.press and when you detect a button release, you should do bleKeyboard.release.

Aieser, 2020-08-02:
Ok, I will try that. Can I use bleKeyboard.release whit alfabet keys? Like 'a'.

T-vK, 2020-08-02:
Please refer to the keyboard documentation as mentioned in the readme. https://www.arduino.cc/reference/en/language/functions/usb/keyboard/
There is massive amounts of information on it on the Internet.
If you have coding issues that are not specific to the ESP32-BLE-Keyboard library such as in this case, https://forum.arduino.cc/ is the apropriate place to ask for help.