T-vK/ESP32-BLE-Keyboard #34: MacOS (10.15.1 (19B88)) "Disconnect" doesn't work from Bluetooth in menu bar

nikolayrantsev, 2020-08-08:
Bluetooth in menu bar -> Devices -> "ESP32 device name" -> Disconnect (only this option is available) 
Clicking on Disconnect here - my device reconnects automatically immediately

[image]
Additional info:
When you go to Apple menu/System Preferences/Bluetooth
only Remove option (not Disconnect) available for the connected device
[image]

T-vK, 2020-08-08:
Unfortunately I don't have any Apple devices, so I can't really help too much. Does this issue occur on other devices too (Windows or Android maybe)?

nikolayrantsev, 2020-08-08:
Hi, 
Thanks for the feedback, 
I've checked on Windows (Bootcamp) - there is no feature like Disconnect - only Remove, 
So, this Apple's specific I guess, 

Btw - iOS device can do only "Forget this device" no Disconnect as well,

T-vK, 2020-08-08:
In that case, I can't really help. Maybe someone else with a Mac OS device can jump in.

nikolayrantsev, 2020-08-08:
Thank you,
Hope, somebody will be interested in this!

nikolayrantsev, 2020-08-14:
Hi again, 
Here are my findings regarding this behavior.
I've flased the sample [esp-idf/examples/bluetooth/esp_hid_device/](https://github.com/espressif/esp-idf/tree/c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b/examples/bluetooth/esp_hid_device) 
Bluetooth connectivity has another behavior:
1. It can be disconnected via the clicking on the option in the menu bar:
[image]
2. However, the device disappears from the menu at all
[image]
3. And can be only removed, not connected again
[image]

I know it's not Arduino ( but ESP-IDF) sample. The purpose was just to check the Connect/Disconnect/Remove options functioning. 

My guess: there are settings in kind of ESP32 core libraries on how to handle the connection. 
Appreciate somebody's effort to dive into the topic and tune this to make it finally ideal! Thanks!

T-vK, 2020-10-20:
Can you try this again with the latest release (0.2.2), please?

nikolayrantsev, 2020-10-21:
Thanks, everyone for the effort!
Name of the device recognizing fine (as defined in the code) the first time you are connecting it on macOS, - fixed/improved.
"Disconnect" still has the same behavior as early - reconnects immediately after clicking on the Disconnect menu item,

T-vK, 2020-10-21:
Thank you for testing this again. I was hoping that the changes may have also had an effect on that reconnecting behavior. I guess not. Hopefully someone in the future can come up with a fix.