T-vK/ESP32-BLE-Keyboard #35: MacOS (10.15.1 (19B88)) /System Preferences/Bluetooth device initially discovers with some factory name

nikolayrantsev, 2020-08-08:
When you're connecting device for the first time after uploading the code and rebooting ESP32, 
it discovers with the factory name, not defined in the code

[image]

When devices paired, it shows with the defined name in the code:

`BleKeyboard bleKeyboard("Swiping");`

[image]

T-vK, 2020-08-08:
I think that's a Mac "feature". If no name is specified, it would fall back to "ESP32 Bluetooth Mouse". What your Mac is showing seems to be a the Mac address with dashes instead of colons. I would assume that it shows the correct name after a while or maybe after reloading the list?

nikolayrantsev, 2020-08-08:
Yeah, the correct name shows after turning Bluetooth on/off

aovestdipaperino, 2020-10-15:
I realized that MacOS seems sensitive to chars.
KBV2 is ok
KBV2.0 is not ok

aovestdipaperino, 2020-10-16:
Created a [PR](https://github.com/T-vK/ESP32-BLE-Keyboard/pull/49)

T-vK, 2020-10-20:
Thank you, I have accepted the PR.

closed 2020-10-20 by T-vK