T-vK/ESP32-BLE-Keyboard #36: With iOS it seems to work perfectly but the iPhone wifi slows down and doesn't work

Harmonium80, 2020-08-15:
I can't understand how. When I pair esp32 with ESP32-BLE-Mouse or ESP32-BLE-Keyboard to an iPhone (I tested it with an 8 and an 8 plus) even in a sketch with only #include <BleKeyboard.h>, BleKeyboard bleKeyboard and bleKeyboard. begin () in the setup the wifi of the phone is very slow, sometimes it stops ... what happens?

T-vK, 2020-08-16:
Honestly, I don't know. Sounds like it could be a driver/firmware issue on the OS side. I guess besides updating your iOS there is no way to change the driver/firmware.
Using a 5Ghz wifi might help because then the Bluetooth and Wifi in the phone don't have to share the same antenna.

Harmonium80, 2020-08-17:
I tried to write on esp32 this https://github.com/neilbags/arduino-esp32-BLE-MIDI/blob/master/BLE_MIDI.ino and when I pair it to the iPhone the problem is not there so I think the fault it is not of the antenna. Anyway I will try to do other tests ...