T-vK/ESP32-BLE-Keyboard #37: Is it incompatible to use CSR Dongle in Win7 or Win10?

LusterWong, 2020-08-17:
Thank you very much for this wonderful library.

A BLE KB test on ESP32 DEVKIT found iphones and Android phones working fine, but I tested three dongle chips using CSR dongle in Windows 7 or 10, and Windows couldn't find bluetooth for ESP32.

I tried installed the CSR driver(CSR Harmony)  test and it also failed.

What should I do, thank you

T-vK, 2020-08-17:
I don't know. Probably depends on if it supports BLE or not.

LusterWong, 2020-08-26:
> I don't know. Probably depends on if it supports BLE or not.

The chip I used was CSR8510 and confirmed that it was BLE enabled, but I've tried five PCs and still can't recognize the Bluetooth keyboard for ESP32.

In addition, I also bought dongle of BCM20702, but it still cannot recognize the Bluetooth keyboard.

beastdjw, 2020-10-09:
Same problem for me. Android,Apple,Linux works fine. But windows not, when I use this library: https://github.com/scottlawsonbc/espjoy , Windows finds the esp32.

LusterWong, 2020-10-12:
> Same problem for me. Android,Apple,Linux works fine. But windows not, when I use this library: https://github.com/scottlawsonbc/espjoy , Windows finds the esp32.
Thank you very much, this can be searched in windows.
But  i don't know how to modify for the HID keyboard program?

T-vK, 2021-08-30:
It might be worth it to try #111 and see if that happens to fix the compatibility issues with that dongle.