T-vK/ESP32-BLE-Keyboard #46: Problem with connected statement

juanhuarte, 2020-09-21:
hi
 so, first of all, I really enjoyed working with the library, it makes everything very clear end easy. but I have a problem with the connected statement. i thinks it sometimes returns true when it really is false.
im working with low power consumption (light sleep mode) and after i wake up the ESP32 with an external button i write an if condition so that if it is connected, it has to return the word "Conectado" . The problem is that sometimes it enters to the if conditional but does not print the word "conectado", so i assume it wrongfully enters into the if command even if its not connected.  Could it be that working with low power consumption is interfering with the code? 
   while(!bleKeyboard.isConnected()){
        Serial.print(".");
        delay(100);
        }
      if (bleKeyboard.isConnected()) {
       bleKeyboard.println("conectado"); // entre 200 y 300 m 
         delay(200);
         Serial.println("entre al conectado");
       }
and also i wanted to ask, which function can i use if i want to unable the bluetooth. Because i tried to use bleKeyboard.end() and it doesnt work. thank you very much.

here is the complete code

#include <BleKeyboard.h>
#include <WiFi.h>
#include <esp_wifi.h>
#include <esp_bt.h>
//Nombre del dispositivo que me va a aparecer en las configuraciones de bluetooth
BleKeyboard bleKeyboard("ESP-32 Keyboard");

struct Button {
  const uint8_t PIN;
  uint32_t numberKeyPresses; //unsigned int 32 bits
  bool pressed;  //two values, true or false.
};
const int ledpin=2;
boolean oldPinState = LOW;
Button button1 = {33, 0, false};

RTC_DATA_ATTR int bootCount = 0;

// define ISR function for PushButtton's Interrupt
void IRAM_ATTR isr() {
 // if(bootCount>1){
   if (oldPinState == LOW) {
         button1.numberKeyPresses += 1;
         button1.pressed = true;
         digitalWrite(ledpin,HIGH);
       
      }
      oldPinState = HIGH;
 // }
}

void print_wakeup_reason(){        // funcion para ver de que modo "despierto la placa" 
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

void setup(){
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  // comienza a aparecer el teclado en los dispositivos 
  bleKeyboard.begin();
  
  pinMode(ledpin,OUTPUT);
  pinMode(button1.PIN, INPUT_PULLUP);
  attachInterrupt(button1.PIN, isr, FALLING);
  ++bootCount;
  //incremento la variable para saber cuantas veces "desperte" la placa
 // ++bootCount;
 // Serial.println("Boot number: " + String(bootCount));
 // pinMode(33,INPUT_PULLDOWN);
    // atiendo la interrupcion cuando presiono el boton
 // attachInterrupt(digitalPinToInterrupt(33), isr, 1);

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

//  rtc_gpio_pulldown_en((gpio_num_t)GPIO_NUM_33);

// despierto la placa cuando el boton pasa de 0v a 3.3v
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_33,0);
//  esp_sleep_enable_ext0_wakeup(button1.PIN,FALLING);
  
 // esp_sleep_enable_ext0_wakeup((gpio_num_t)GPIO_NUM_33,RISING); 

  //Go to Light sleep now
  Serial.println("Going to light sleep now");
  delay(1000);
//  WiFi.forceSleepBegin();
// WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
 // btStop();
//  esp_wifi_stop();
//  esp_bt_controller_disable();
  esp_light_sleep_start();
  Serial.println("me desperte");
 // chequeo que este en sleep mode
 if (bleKeyboard.isConnected()) {
  bleKeyboard.print("This will never be printed");
 }
}

void loop(){
  
  if (button1.pressed) {
      Serial.printf("Button 1 has been pressed %u times\n", button1.numberKeyPresses);    
      digitalWrite(ledpin,LOW);
      while(!bleKeyboard.isConnected()){
        Serial.print(".");
        delay(100);
        }
      if (bleKeyboard.isConnected()) {
       bleKeyboard.println("conectado"); // entre 200 y 300 m 
         delay(200);
         Serial.println("entre al conectado");
       }
      Serial.println("Going to Light sleep now after printing de leter");// ver bien el tema del delay y del serial
      button1.pressed = false;
      delay(500);
      oldPinState = LOW;
       esp_light_sleep_start();
       Serial.println("me desperte"); 
       delay(1000);
  }
}

T-vK, 2020-09-21:
To disable Bluetooth you could try the deinit method I mentioned here: https://github.com/T-vK/ESP32-BLE-Keyboard/issues/43#issuecomment-687639347
I haven't actually tested it though.

I don't have an explaination for why the isConnected method returns true before you can actually send anything. Maybe you need to give it some time after it connected before you can send anything.
I would avoid while-loops like that and also avoid delays and instead write non-blocking code. This may already solve your issue. https://randomnerdtutorials.com/why-you-shouldnt-always-use-the-arduino-delay-function/

DustinWatts, 2020-11-16:
This method of disabling Bluetooth works for me:
```
btStop();
esp_bt_controller_disable();
esp_bt_controller_deinit();
esp_bt_controller_mem_release(ESP_BT_MODE_BTDM); 
```

This doesn't delete the FreeRTOS task created by bleKeyboard though. I have a pull request that makes bleKeyboard.end() do that. I do understand that in a future version that would not be necessary anymore, but I could be a solution for now.

mderrick, 2023-01-08:
I can confirm this happens to me also. `isConnected()` is true and the message is not sent. I'm in a similar situation trying to awake from deep sleep, and send a message as quickly as possible, a delay in this doesn't work for my use case. The example in the readme seems to work for me because of the 5000ms delay between `bleKeyboard.isConnected()` checks.