T-vK/ESP32-BLE-Keyboard #47: Configurable media keys?

bilkusg, 2020-09-22:
Firstly, many thanks for the library - I've been using it to set up an ESP32 to convert infrared keycodes into keyboard input to an Amazon FireStick 4K so my wife can use a firestick with the remote she likes.

The only difficulty I've had is that the firestick needs to receive a specific media key for its menu button, and the only way I've found of achieving that is to physically edit the bleKeyboard.cpp file to change the media key mapping for one of the 16 assigned keys.

That works fine, but it's ugly and won't make it easy to track future changes to the library.

So, my question is: would it be worth adding a new  API to be called before begin() which allows the media key assignments to be overridden? It's clearly not a vast amount of work, and I'd be happy to volunteer to do it if there were sufficient interest.

zephyrr, 2020-11-02:
I'm trying to understand this; please excuse me if this is too naive.

You are changing for different media codes by editing bleKeyboard.cpp; I'm not sure where you find appropriate codes to edit in (definitions), if I wanted to edit for different codes (even if I wanted to edit the cpp file while awaiting the configurablity you are discussing).

I see codes bleKeyboard.cpp lines 72+ define some one byte codes with comments associating them with media key actions.
So for example there is B7 for Stop, and CD for play/pause.

Where is the meaning of these codes defined?

I see for example different media key codes described in:

Like 0x55 for play/pause and 0x56 for stop

Where should I be looking?

(My project - to replace a cheap commercial media remote button for controlling podcast playback on my android phone.  The current remote button is powered by a coin cells, to it quickly sleeps and loses connection, then has to re-establish BT after wakning up, and may miss a few commands.  Also toggling play/pause sometimes is ignored, so pressing it again is needed - but then sometimes after a delay it sees both and toggles back.  Very frustrating.  I'd like to send unambiguous PLAY or PAUSE, with separate buttons, rather than a state dependent toggling PLAY/PAUSE code invoked with a single button.

So I'd like to power the ESP with a larger battery so it can maintain a continuous bluetooth pairing while we walk, and also send unambiguous commands with separate buttons.  But I don't understand which codes to send - whether through editing the cpp file or later with some kind of configuration.)

Guillemete, 2023-05-23:
did you have any luck?

i manage to send separate PLAY and PAUSE modifiying the CPP file 
  USAGE(1),           0xB1,          //   USAGE (Pause)               ; bit 3: 8
  USAGE(1),           0xB0,          //   USAGE (Play)                ; bit 4: 16

works on iOS devices, but not on pc, or android.