T-vK/ESP32-BLE-Keyboard #48: [iOS] New Line seems not to be send corectly

SunboX, 2020-10-06:
If I'm trying to send something like this:

```
bleKeyboard.print("0000111133335\n");
bleKeyboard.print("0000111133335\n");
bleKeyboard.print("0000111133335\n");
bleKeyboard.print("0000111133335\n");
```

it mostly gets printed on iOS like this:

```
0000111133335000011113333500001111333350000111133335
```

and sometimes like this

```
0000111133335000011113333500001111333350000111133335

...
```

T-vK, 2020-10-07:
Try println: https://www.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardprintln/
Or check: https://forum.arduino.cc/index.php?topic=243947.0

SunboX, 2020-10-07:
Thx, will try it!

T-vK, 2020-10-07:
The example has an example for sending an enter key as well: https://github.com/T-vK/ESP32-BLE-Keyboard/blob/master/examples/SendKeyStrokes/SendKeyStrokes.ino

SunboX, 2020-10-08:
Sadly, none of both versions did work. I tried this one:

```c++
bleKeyboard.println("0000111133335");
```
and also this one:
```c++
bleKeyboard.print("0000111133335");
bleKeyboard.write(KEY_RETURN);
```

T-vK, 2020-10-08:
Can you try adding delays and separating the press and release for the ENTER key?

Like this:

```
bleKeyboard.print("0000111133335");
delay(500); // wait 1/2 a second
bleKeyboard.press(KEY_RETURN); // Hold enter key down
delay(500); // wait 1/2 a second
bleKeyboard.release(KEY_RETURN); // Release enter key
delay(500); // wait 1/2 a second
```

Do other characters work? What about letters that involve the shift key? Can you test `bleKeyboard.print("Hello");`?

SunboX, 2020-10-08:
Yep, that did the trick:

[image]
[image]

SunboX, 2020-10-08:
I did some tests, 30ms has many missed line breaks, 40ms still has some occasional misses, with 50ms it seems to be fine.

SunboX, 2020-10-08:
Can we add some config parameter for such a delay between key events?

T-vK, 2020-10-08:
I want to keep the BLE Keyboard library compatible with the USB Keyboard library, so probably not. But feel free to write your own function that wraps the print, press, release and delay.

T-vK, 2021-08-30:
We have added a default delay of 7 milliseconds in PR #111 and also added a `setDelay` method. We need some people to test it before we can merge it and release a new version though. (Especially with Mac OS and iOS devices.) Maybe you can give it a try?