T-vK/ESP32-BLE-Keyboard #51: List of supported / unsupported Apple devices (iPhones, iPads, Macs etc.)

T-vK, 2020-10-21:
I would like to maintain a list of Apple devices here that work or don't work with this library.
If you have tested a device that is not listed with the same OS version yet, please add a comment and tell me which device and operating system (i.e. iOS version or Mac OS version).
If you tested a device that is already listed, but you had different results (e.g. list says it doesn't work, but for you it does work), please add a comment as well. (And don't forget to mention your device name and OS version.)

**Devices that work**

- [x] iPhone SE | iOS 12.3.1 | Reported by [satoer](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/2#issue-482741158)
- [x] iPad Air 2 | iOS 12.3.1 | Reported by [satoer](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/2#issue-482741158)
- [x] iPad Pro 2 | iOS 12.3.1 | Reported by [satoer](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/2#issue-482741158)

**Devices that don't work**

- [ ] iPad 2 | iOS 9.3.5 | Reported by [satoer](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/2#issue-482741158)

blackketter, 2020-10-21:
A good rule of thumb would be that any device that doesn't support Bluetooth 4.0 or higher would not be compatible.  Bluetooth 4.0 is where Bluetooth Low Energy, (aka Bluetooth LE, aka BLE, aka Bluetooth Smart) was introduced into the standard.  

For Apple products, Bluetooth 4.0 or 4.2 was introduced with the: 3rd generation iPad in 2012, iPhone 4s, MacBook Air 2011, MacBook Pro 2012, MacMini 2011, iMac 2012.

I think that having a list (maybe in the top README) of tested devices (Apple and not) is valuable in any case.

T-vK, 2020-10-22:
Thanks, good point, I will add that to the list.
Yeah I was planning to add a link to this issue in the README where it says "Compatible with iOS".

iwbrhwfy-source, 2020-11-04:
Library version 0.22 iPad Air 10.3.3 and iPhone 7+ 12.x.x can't find device
Library version 0.21 iPad Air 10.3.3 and iPhone 7+ 12.x.x can find device and connect, but very limit by app (test ok with text in messenger and Page Turner in Piascore)

T-vK, 2020-11-04:
Thanks for reporting. Can you explain what you mean by "very limited by app"?

iwbrhwfy-source, 2020-11-06:
> Thanks for reporting. Can you explain what you mean by "very limited by app"?

you can't scroll image on Photo app by send LEFT ARROW, RIGHT ARROW key, but i test on iPad gen 6 or 7 it work OK

T-vK, 2020-11-06:
I think the issue is with the app in that case. Maybe the app simply doesn't support scrolling using the arrow key. You could try using [ESP32-BLE-Mouse](https://github.com/T-vK/ESP32-BLE-Mouse) for that instead.

iwbrhwfy-source, 2020-11-06:
> I think the issue is with the app in that case. Maybe the app simply doesn't support scrolling using the arrow key. You could try using [ESP32-BLE-Mouse](https://github.com/T-vK/ESP32-BLE-Mouse) for that instead.

thank you i will try :D my ipad too old :(

frank7fan, 2020-11-23:
Tested with v0.2.2, works with Windows 10 and Android 7.0, but NONE of my Apple devices can see the advertisement from ESP32:
iPhone 6, iOS 12.4.6
iPhone 11, iOS 14.2
iPhone 12, iOS 14.1
MacBook Pro Mid 2012, 10.13.6
MacBook Pro 2017, 10.14.6
iPad 2, iOS  9.3.5
iPad 4, iOS 10.3.3
iPad Air 3rd Gen, iOS 14.2

T-vK, 2020-11-23:
Imteresting can you try release 0.2.2 again but delete this line? https://github.com/T-vK/ESP32-BLE-Keyboard/blob/928c09320e3e6e344a63640c7de21d52a20c1d47/BleKeyboard.cpp#L152
When you say "can find the ESP32", do you just mean find or can you actually connct and send keystrokes successfully?

frank7fan, 2020-11-23:
- Release 0.2.1: can see and can connect with all devices. I only tested with volume up key and they all work.

- Release 0.2.2:
1. No code change and use "BleKeyboard bleKeyboard;" (default device name): No Apple device can see the ESP32.
2. Remove line 152 of BleKeyboard.cpp **OR** use "_BleKeyboard bleKeyboard_("ESP32-Keyboard2", "Me", 100);":
iPhone 6, iOS 12.4.6: can see, can pair, can send Volume up key
iPhone 11, iOS 14.2: can see, can pair, can send Volume up key
iPhone 12, iOS 14.1: can see, can pair, can send Volume up key
MacBook Pro Mid 2012, 10.13.6: can see, cannot pair (jump between connect and disconnect)
MacBook Pro 2017, 10.14.6: can see, can pair, can send Volume up key
iPad 2, iOS 9.3.5: need turn off and on BLE before can see, can connect, can send Volume up key
iPad 4, iOS 10.3.3: need turn off and on BLE before can see, can connect, can send Volume up key
iPad Air 3rd Gen, iOS 14.2: need turn off and on BLE before can see, can connect, can send Volume up key

SunboX, 2020-11-25:
@frank7fan Same for me, thanks for the hint how to fix it!

T-vK, 2020-11-25:
Maybe this is an issue with the Apple device remembering the device and getting confused because the Bluetooth MAC address didn't change, but the device name did.

I honestly don't know what to do if you say that compatibility for some devices breaks when using 
`pAdvertising->setScanResponse(false);`
but compatibility for other devices breaks when not using it.

Maybe someone can test what I mentioned here again: https://github.com/T-vK/ESP32-BLE-Keyboard/issues/51#issuecomment-731988187

But this time completely disconnect, unpair and forget the ESP32 on each Apple device first.

SunboX, 2020-11-25:
> Maybe this is an issue with the Apple device remembering the device and getting confused because the Bluetooth MAC address didn't change, but the device name did.

No, not really. I can change the name to "FooBar" and it still works. But not "FooBarFooBarFooBar".

frank7fan, 2020-11-25:
Confirmed when device name greater than 15 chars, iDevices will not see the advertising. 
iDevices generic BLE scanner seems keep a table of scan result that has MAC and device name. If a MAC exists in the table, it will skip that device and not update the information (device name) in the following scan. So, I can program ESP32 with device name less than 15 chars and be found by iDevices. Without connection or connected and forget, I program ESP32 with device name more than 15 chars and I can find the device on the list with old device name (less that 15 chars) and I can connect to it. This table will be erased if I turn BLE OFF. So, not to be confused by the test result, I need to run all my tests with setting forget and turning BLE OFF and ON on iDevices.
For the 15 chars limit issue, I tried with nRF Connect (Nordic iOS App Tool). No problem seeing long device name and connecting with keyboard sending function. So, I guess this issue is related to how Apple implement and handle generic BLE scan function.

SunboX, 2020-11-25:
I don't know the BLE spec, but a short Google around tells me 31bytes for device name. UTF8 uses 2 byte per char, or? So it could handle 15 chars and 16 will break the spec. Would make sense. But I'm not sure, because I'm no expert in this.

T-vK, 2020-11-25:
I don't think there is a limit to the device name's length directly. I think the advertising packet has a total size limit and the max device name length is dependent on what else is in the packet. But I'm no expert on this topic either...

So now that we figured out that having a long device name causes issues, it would be nice to see some clean tests (disconnect, unpair, forget, clear cache by turning off Bluetooth before every test) comparing the unmodified library vs the one without
`pAdvertising->setScanResponse(false);`.

robomaniac, 2020-12-09:
I can confirm this code works with

- iphone12 pro iOS 14.2.1
- iPhone 8 iOS 13.7
- Samsung S9 Android 10

This takes pictures on all 3 phones. I can not work on my 3D printer timelapse!!

```
/*

Tested on
iphone12 pro iOS 14.2.1
iPhone 8 iOS 13.7
Galaxy S9 Android 10 

with LILYGO esp32 TTGO with OLED

@Title BLE keyboard, arcade button to phone
@Author Geert Roumen
@Institute Umeå Institute of Design
@Date 29-01-2020
 
@Hardware
 
Any ESP32 dev board would do, but I used the:
Geekworm EASY KIT ESP32-C1
https://geekworm.com/products/official-geekworm-esp32-development-board

Arcade button
Adafruit Large Arcade button white
https://www.adafruit.com/product/1192

D13       Button pin
3.3V      Connected to the other leg of the button
 
In this example sketch processing can send an RGB value to Arduino and Arduino will change the LED color of the NUMPIXELS connected neopixels
*/
#include <BleKeyboard.h>
//Se the name of the bluetooth keyboard (that shows up in the bluetooth menu of your device)
BleKeyboard bleKeyboard("shutter");

const int buttonPin = 0;
//Set the old button state to be LOW/false; which means not pressed
boolean oldPinState = LOW;

void setup() {
  //Start the Serial communication (with the computer at 115200 bits per second)
  Serial.begin(115200);
  //Send this message to the computer
  Serial.println("Starting BLE work!");
  //Begin the BLE keyboard/start advertising the keyboard (so phones can find it)
  bleKeyboard.begin();
  //Make the button pin an INPUT_PULLDOWN pin, which means that it is normally LOW, untill it is pressed/ connected to the 3.3V
  pinMode(buttonPin, INPUT_PULLUP);
}

void loop() {

  if (bleKeyboard.isConnected()) {
    //if the keyboard is connected to a device
    if (digitalRead(buttonPin) == LOW) {
        Serial.println("Cheese!");
      //If the button pin is pressed (since it is pulled down, it is pressed when it is high
      if (oldPinState == HIGH) {
        //if the old Pin state was LOW and the button pin is HIGH than...
        //send the spacebar
       //bleKeyboard.print("j");
        //or you can comment this one out, and it will send the newline character
        bleKeyboard.write(KEY_MEDIA_VOLUME_UP);
      }
      oldPinState = LOW;
    } else {
      oldPinState = HIGH;
    }
  }
  delay(300);
}
```

armantaherian, 2020-12-22:
I just test the default example with Apple MacBook Pro 2020 with M1 chip, and I am not able to find the device, then also I tested it with MacBook Pro Mid 2012 and also there I can't find it, but when I upload BleMouse example on my esp32 and pair it with those MacBooks, when the device paired I'm able to upload BleKeyboard example on the esp32 and everything works fine, but if I unpair the esp32 again I can't find it!

LeeNX, 2021-01-27:
On a macOS Catalina Version 10.15.7 - MacBook Pro (15-inch, 2017), I can't see any unbound ESP32-BLE devices from the OSX BlueTooth Bar or App, I need to use an application called LightBlue LE to see all BLE devices and force a connection to my ESP32-BLE device and then connection happens each time the device is on. I have not tested with shorter device names. Not sure if that helps.

yepher, 2021-02-24:
Using [`v0.2.2`](https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.2.2) 

* MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports) does the device in the list. But I can connect if I force connection like @LeeNX mentions
* iPad Pro 9.8-Inch (MLQ26LL/A) does not the device in the list but works fine if I force the connection 
* iPhone 6s (MKR32LL/A) Did not try and connect but did not see the device in the list

I commented out BleKeyboard.cpp LN:152 `//pAdvertising->setScanResponse(false);` and I see the advertisement and can see the ESP32 on all the devices mentioned above

jokkebk, 2021-05-13:
Couldn't see the device in iPad Pro (2nd generation, BT 4.2) when initialized with:

BleKeyboard bleKeyboard;

Works great with suggested line:

BleKeyboard bleKeyboard("ESP32-Keyboard2", "Me", 100);

mzimm, 2021-05-17:
Just replicated @frank7fan's suggestion above. Thanks! 
substituting in:
BleKeyboard bleKeyboard("ESP32-Keyboard2", "Me", 100);  
Works.
So many posts across the web saying ESP32 connection to ios not possible I was about to give up.
ESP32 Feather Huzzah (Adafruit)
iPad Pro (V 1) running iPadOs 14.5.1 can now see and pair with ESP32.
Sent text to 'notes' app using the basic example. Also tested my real goal
of sending keyboard shortcuts to Procreate. Sending 'b' switched from eraser to
the brush tool. Oh, Yeah. 
Older iphone SE running ios 13.7 cannot see the device.
Interestingly now iPad pro seems to be looking for the ESP32.
(and maybe was already? when the iPad initially found the updated ESP32 it had the older name from the 
example code I am using "Arcade Phone"..???)
Also, changing the line back to:
BleKeyboard bleKeyboard("ESP32-Keyboard2"); now works as well.

kotobuki, 2021-05-22:
@T-vK First of all, thank you very much for creating such a useful library!

I just confirmed that the following devices worked with limiting the device name within 15 characters as @frank7fan [suggested](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/51#issuecomment-733886764)

Mac mini (M1, 2020) | macOS Big Sur 11.3
iPhone XS | iOS 14.5.1

nanosonde, 2021-05-28:
An advertising packet can be up to 31 bytes of data. Each field (i.e. appearance, name, service UUID or similar) have a header of 2 bytes (length and type), meaning that the maximum user payload is 29 bytes.

When you reach this limit, the ble_advdata_set() method of the SDK will start returning an error, NRF_ERROR_DATA_SIZE (0x0C).

Remember that if you need to transfer additional data before establishing a connection, you can add a scan response, which can contain an additional 31 bytes. A scan response packet can be set by passing a pointer to an addtional ble_advdata_t as the second parameter of ble_advdata_set().

(Origin: https://devzone.nordicsemi.com/f/nordic-q-a/14/what-s-the-maximum-size-for-an-advertisement-package)

Leave aside the fact that it talks about the NRF SDK. The size limitation is according to the BLE spec.

See also this: https://jimmywongiot.com/2019/08/13/advertising-payload-format-on-ble/

ShivamJoker, 2021-08-24:
To all the ones who are struggling to get it show in the latest MacOS

Add this at initialisation
```
BleKeyboard bleKeyboard("ESP key", "ME", 100);
````

And viola 
[image]

T-vK, 2021-08-30:
#111 Should fix most of the iOS and Mac OS issues once you disconnect and unpair/forget the ESP on your device and then reconnect. The default name has been changed to `ESP32 Keyboard` and if you specify a longer name it will only use the first 15 characters. Btw the fix also greatly reduces the memory footprint allowing the parallel use of Wifi and Bluetooth.
We still need some people to test it though before we can make a new official release. So if you could give it a try and report back that would be highly appreciated. Ideally including the model name of your device and the iOS version / Mac OS version.

ShivamJoker, 2021-08-31:
@T-vK how can we test that release ? I am using platformio

T-vK, 2021-08-31:
I guess the easiest way would be to download that branch as a zip file and install that just like you would do with a normal release.
https://github.com/sivar2311/ESP32-BLE-Keyboard/archive/refs/heads/master.zip

sivar2311, 2021-08-31:
The easiest way to test this using PlatformIO is to edit the platformio.ini:
```ini
lib_deps =
  https://github.com/sivar2311/ESP32-BLE-Keyboard
```

or for NimBLE-Mode:
```ini
build_flags=
  -D USE_NIMBLE

lib_deps =
  NimBLE-Arduino
  https://github.com/sivar2311/ESP32-BLE-Keyboard
```

tahunus, 2021-09-08:
Worked with iPad Pro 9.7 (2016) and IOS 14.7.1 when using BleKeyboard bleKeyboard("ESP32Kbd") with library version 0.2.3

iOS issues this warning: "Using ESP32kbd may affect Wi-Fi and Bluetooth connectivity"

My board is an Adafruit HUZZAH32 (ESP-WROOM-32) (WiFi + BLE)

Longer keyboard names or names with spaces or special characters do not appear on iOS´s Bluetooth devices list (i.e. ESP32-BLE-Keyboard or ESP32 BLE Keyb)

ShivamJoker, 2021-09-10:
Okay @T-vK @sivar2311 this seems to fix the issue.
Thanks a lot 
[image]

T-vK, 2021-09-10:
Okay great, so it works on Mac OS. :)

Is anyone with an iOS device willing to test that version? https://github.com/sivar2311/ESP32-BLE-Keyboard/archive/refs/heads/master.zip

kotobuki, 2021-09-10:
I just confirmed with my iPhone XS (iOS 14.7.1) with both default and a long name. Both worked fine, and the long name was truncated at 15 characters as expected.

```
BleKeyboard bleKeyboard("Loooooong012345678901234567890");
```

T-vK, 2021-09-10:
Thank you, I've created an official release including all these fixes now. [v0.3.0](https://github.com/T-vK/ESP32-BLE-Keyboard/releases/tag/0.3.0)

barrydegraaff, 2021-10-31:
I can confirm the latest release works on IOS15.0.2 using iphone se2 using custom device name. Actually typed this comment using a ESP32 keyboard.

DESECTORCON, 2022-08-03:
Working on

- Ipad pro 2018 model running on IOS15.5 with line:
BleKeyboard bleKeyboard("DockingBay01", "SEIS", 100);

Instantly found and connected :)

tambourine-man, 2023-03-13:
“iPad 2, iOS 9.3.5: need turn off and on BLE” 

Hi @frank7fan, can you expand on that? How did you turn off and on BLE on the ESP32? 
Can you confirm that you were able to connect with iPad 2?  Everywhere I searched claims it can't be done with this device. 
Tried eveything in this thread and nothing worked.
Thanks a lot

carburator03, 2025-04-21:
The v0.3.0 works great with an iPhone 12 (iOS versions 17.x and 18.x). The device appears immediately in the Bluetooth settings, even with custom names.

However, it's a bit tricky to get it working with an iPhone 15 (and probably later models running iOS 18.x). It seems Apple considers it less secure, so the device doesn’t appear in the Bluetooth settings at all. But I found a workaround:
Go to **Settings -> Accessibility -> Switch Control -> Switches -> Bluetooth Devices...** and there it is, ready to pair. You only need to navigate there the first time when pairing. After it's connected, go back to the Bluetooth settings, now it will be listed there. It reconnects and works flawlessly, just like with the iPhone 12.

Hope someone finds this helpful!