T-vK/ESP32-BLE-Keyboard #55: Control+keys a-z not working

dashiad, 2020-11-14:
Hello! Thank you for sharing the project, i'm using it to add bluetooth support to an IBM Model M, but using it with Windows, combinations of keys CTRL+<a-z>, are not sending bluetooth reports. 
The following code:

` 
VirtualKey virtualKey = keyboard->getNextVirtualKey(&keyDown);
uint8_t keyCode = convertToKeyCode( virtualKey);
    if (keyDown)
    {
       
        bleKeyboard.press(keyCode);
        
    }`
for <Ctrl-a> , gets a VirtualKey of 22 (0x16), which is converted to keyCode 1, which is used as an index for the _asciimap array in BleKeyboard.cpp, and the value for that index is 0, so it's discarded by the press() method:
`if (!k) {
			setWriteError();
			return 0;
		}`

Changing the asciimap array so the offsets 0x01,0x02...have values of 0x04,0x05... doesnt really work, as it then confuses the key combination Ctrl-H with the backspace character, So, finally, i changed the virtualKeyToASCII so VK_a ...VK_z  and VK_A ... VK_Z returns a keycode independent of CTRL being pressed or not. 
So,pressing H makes convertToKeyCode return a 0x1D, independently of CTRL being pressed or not. This is correctly interpreted by Windows (maybe would cause problems in other scenarios?), as CTRL-H, different from backspace.

T-vK, 2020-11-14:
Are you sure this is a problem with ESP32-BLE-Keyboard? 
Let's take everything else out of the equation. Try running this code:

``` C++
#include <BleKeyboard.h>

BleKeyboard bleKeyboard;

void setup() {
  bleKeyboard.begin();
}

void loop() {
  if(bleKeyboard.isConnected()) {
    bleKeyboard.press(KEY_LEFT_CTRL);
    delay(100);
    bleKeyboard.press('a');
    delay(100);
    bleKeyboard.releaseAll();
  }
  
  delay(5000);
}
```

dashiad, 2020-11-14:
Opps, sorry, i reported this in the wrong repo...This is indeed a problem with a project using this library (https://github.com/abelykh0/esp32-BLEKeyboard), not with the library itself.
Sorry for the confusion!

closed 2020-11-14 by dashiad