T-vK/ESP32-BLE-Keyboard #56: media controls issue when disconnected-reconnected

Frost159, 2020-11-16:
Hi, I have used this library a lot in a macro bluetooth keyboard project, but I have an issue with using media controls. When I first connect my esp32 with a device (for example my smartphone) I pair it and it works (all bluetooth keyboard keys are being sent, including media keys). but when I disconnect my esp32 from the device, and I reconnect it, all the keyboard keys will work as expected, however media keys will no longer work. The only way I found to make media keys work again is to unpair my esp32 from my device by deleting it from the device list, then pair it, and it will work again.
I have had this issue with my full program but also with the example program of the library. I have spoke with other people using this library for similar projects, some had the same issue, others don't.
I wonder if it is an issue based on the disconnection between the esp32 and the device that is not proprely made, or if it's an issue with a buffer.
Anyway this library has still be very usefull for any other keyboard keys and it has helped me a lot for this project, Thanks.

Frost159, 2020-11-16:
I have not mentionned it, but I'm using a [ttgo t display esp32 board](https://github.com/Xinyuan-LilyGO/TTGO-T-Display) , maybe the issue comes from the board.

T-vK, 2020-11-17:
I don't know what could be causing this. Does this only happen on one of your devices or can you reproduce this issue with any device? What kind of device did you test against btw? (Android, iOS, Windows, Mac OS, Linux, ... ?)

Frost159, 2020-11-17:
I had this issue with my computer (windows 10) and my android device (galaxy S10)

songotag, 2021-03-13:
I've the same issue : when my phone is connected for the first time to the esp32 module with the BleKeyboard driver works fine. As soon as the phone is disconnected, I can no longer connect, I have to restart the ESP32 for it to work again.

evg26, 2021-03-30:
I've the same issue

zalexzperez, 2021-04-03:
Same here.

Let's say we have phone1 and phone2.

It connects well for the first time on phone1.  But if you then turn off phone1's bluetooth and turn it back on, the connection is not restablished. 

Actually, the ESP32 bluetooth keyboard device is no longer visible to phone2 (other devices) right after phone1 made the connection. This makes me believe the issue happens once the device connects for the first time, not after the disconnection takes place.

As others have said, the Bluetooth connection is only restablished after ESP32 is rebooted. 

This is terrible, as I'm powering the ESP32 directly from a headunit USB connection and my car shuts the electronics off while the car is igniting. 

Android 11 phones.

jbkuma, 2021-04-08:
I think the issue here is that after the connection is broken by whatever means (disconnect, BT off, computer restart) the ESP32 isn't put back in advertising mode.  I was tinkering around with things like bleKeyboard.end();  bleKeyboard.begin();, but .end() is actually an empty function that doesn't do anything.  

I think ideally what we need is a method to start advertising after the device is disconnected.  This doesn't need to be automatic, we can manually trigger it with bleKeyboard.isConnected().

I think if possible, an isAdvertising() function would help with some other house keeping.

I'll keep tinkering and see if I can figure out how to do this myself, but this nuts and bolts are a bit over my head.

T-vK, 2021-04-08:
Take a look at #53. The PR implements a startAdvertising method. The reason I haven't merged it yet is that you have to manually call it, breakinng backwards-compatibility. I someone comes up with a way to automate this eventually. 
The end method is empty indeed for now. I think I just included it because the USB keyboard has an empty end function as well.

zalexzperez, 2021-04-08:
Okay, excuse my ignorance, I'm not a developer, but how exactly can we call it manually and where?

jbkuma, 2021-04-08:
I tested this and it works great, thanks! 

I used Rousir's version Without changing anything in my program, I was able to connect to the ESP32 automatically from a previously paired device, and to find the freshly booted ESP32 from an unpaired device.  I have a button to press to restart advertising, and that works well.

I think it would be safe to merge unless there is another use case that I'm missing.

Rousir closed the PR for some reason, if you aren't ready to merge, maybe you should re-open it so others are more aware of it.  The example sketch he made doesn't seem quite right, but I haven't tested it.  It seems to me it would never retrigger the advertising since the check is within the isConnected() loop.

jbkuma, 2021-04-08:
You need to download the version in the pull request linked above.  Then
you can either manually call bleKeyboard.startAdvertising(); or if you want
it to be automatic use a boolean like bleAdvertising, check isConnected(),
if false and you aren't already advertising start, and set your flag true
so you aren't constantly "starting." On isConnected change your
bleAdvertising flag back to false.

There may be a method to check for advertising but it is not implemented in
the master branch or the #53 pull request.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

T-vK, 2021-04-21:
I hope this got fixed in #86 which is included in the new release (v0.2.3).