T-vK/ESP32-BLE-Keyboard #57: Send Windows key

gitolicious, 2020-11-18:
Does anyone have an idea why we can't send the Windows key? Key codes should be `0x5B` and/or `0x5C` (according to https://stackoverflow.com/a/36458787)
This line does not have any effect on my connected Windows machine - other buttons work fine.

    bleKeyboard.write(0x5B);

DustinWatts, 2020-11-19:
You could try:

`bleKeyboard.write(0x83);`
or
`bleKeyboard.write(0x87);`

gitolicious, 2020-11-19:
Thanks for the hint! `KEY_LEFT_GUI` in fact does somthing _similar_ to the native Windows key on my keyboard. Combining it with `KEY_LEFT_ARROW` moves the active window to the left, but shows a tile preview of my other running applications on the right part of the screen. This does not happen when I use the "real" Win key. There, the windows simply gets moved left. Not an issue really, but shows that something is different.

Also your code sample does not compile as `write()` expects a char, not a string. Replacing it with `write('R')`  does not open the _"Run"_ window as it would if I hit `Windows+R`. It does send the letter `R` though.

Any further hints would be appreciated, even though I circumvented the issue for my specific problem by changing the hotkey assigned to the task I wanted (muting cam and mic with [PowerToys](https://github.com/microsoft/PowerToys/wiki/Video-Conference-Mute-Overview)).

DustinWatts, 2020-11-19:
I made a little mistake when posting the code above. Try this:

```
bleKeyboard.press(KEY_LEFT_GUI);
bleKeyboard.print("r");
bleKeyboard.releaseAll();   
```

_Note: I edited https://github.com/T-vK/ESP32-BLE-Keyboard/issues/57#issuecomment-730201009 to reflect the code in this post._

gitolicious, 2020-11-21:
I thought I tried all combinations and variants - must have missed that one. Thanks a lot! It is working fine now.
Before closing this issue: Would it be helpful to others to mention in the docs that `KEY_LEFT_GUI` corresponds to the Windows key, or even add an alias `KEY_WINDOWS_LEFT`?

T-vK, 2020-11-21:
I don't have docs because I specifically designed the library to be a drop-in replacement for the official Arduino USB Keyboard library.

gitolicious, 2020-11-22:
Understood. With that knowledge I also found [a similar question in the Arduino forum](https://forum.arduino.cc/index.php?topic=391915.msg2974839#msg2974839). It should be documented in the [Arduino keyboard docs](https://www.arduino.cc/en/Reference/KeyboardModifiers) then.

Thanks again for your help. I will close this issue and hope others with the same issue will find the solution here.

closed 2020-11-22 by gitolicious