T-vK/ESP32-BLE-Keyboard #58: iOS14.1 and iOS12.4.8 cannot find ESP32 BLE keyboard

WebSamuraiDaisuke, 2020-11-19:
I use M5StickC have Code include BleKeyboard.h. But they cannot find the device as  ESP32 BLE keyboard.
How iOS find the device with BleKeyboard?

T-vK, 2020-11-19:
Which iOS version and which device and model are you using? Some are not compatible.

WebSamuraiDaisuke, 2020-11-19:
2020年11月20日(金) 3:32 T-vK <notifications@github.com>:

> Which iOS version and which device and model are you using? Some are not
> compatible.
>

iPhone 7 iOS 14.1
iPad 4 iOS 12.4.8

I knew you are the very genius programmer. I’d like you to make it to
compatible for them. If you can make it I’d like you to give donation to
you.

Best regards
Kai

> --
Daisuke Kai Junkai Sent from iPhone

WebSamuraiDaisuke, 2020-11-19:
2020年11月20日(金) 3:32 T-vK <notifications@github.com>:

> Which iOS version and which device and model are you using? Some are not
> compatible
>
iPhone SE2 iOS 14.1 cannot find ESP32-BLE-Keyboard, too.

> --
Daisuke Kai Junkai Sent from iPhone

T-vK, 2020-11-19:
If you look at the list in #51 you can see that the iPhone 7 is not compatible. 
And the iPhone SE 2nd Generation is compatible.

I don't know about the iPad 4. It may or may not work.

Please test it with the official example code that doesn't contain any M5 libraries.

uolot, 2020-11-20:
> Please test it with the official example code that doesn't contain any M5 libraries.

Just to add my 2c, I've been using this lib on M5Stack and M5Atom and using M5 libs, doesn't seem to affect BLE keyboard function at all. Thank you @T-vK for making and maintaining this great piece of software by the way! :)

WebSamuraiDaisuke, 2020-11-22:
@uolot 

> Just to add my 2c,

What is your 2c?

mbrunton22, 2020-11-22:
Yeah, I am also having issues on an iPhone 8 running 14.1 - just cannot see the advertisement (esp wroom 32)

uolot, 2020-11-23:
> @uolot
> 
> > Just to add my 2c,
> 
> What is your 2c?

https://en.wikipedia.org/wiki/My_two_cents

WebSamuraiDaisuke, 2020-11-23:
@T-vK, Will you make this library to be compatible with iOS and macOS in near future?

T-vK, 2020-11-23:
It already is. Again, look at #51. Apple devices that don't properly support "Bluetooth Low Energy" will never work with this and there is nothing I can do about it.

SunboX, 2020-11-25:
Same for me, but it worked with this code here some time ago (same Mac same IOS device): https://gist.github.com/manuelbl/66f059effc8a7be148adb1f104666467

I have to re-test if this is still the case or if the Mac / iOS update also broke that snipped.

mbrunton22, 2020-11-26:
> Yeah, I am also having issues on an iPhone 8 running 14.1 - just cannot see the advertisement (esp wroom 32)

I actually got this working by making my device name really short (in my case I just set it to "Matt"), and it seemed to work after that on both my recent Ipad and an iphone 8 running IOS 14.

WebSamuraiDaisuke, 2020-12-16:
Sorry for my late reply. 
I had to make the device name shorter, every iOS can detect the device.

closed 2020-12-19 by WebSamuraiDaisuke