T-vK/ESP32-BLE-Keyboard #60: Adding License to project?

anujdatar, 2020-11-30:
Would you be willing to add a license to your project? I wanted to incorporate your implementation of media keys in the standard Arduino keyboard library, to use with a matrix keypad. This would be fine for a personal project like mine. But, I also would like to submit a pull request to the Arduino Library with these changes.

I don't think I can do that without a license published in your repository.

Thanks.

anujdatar, 2020-11-30:
If I understand correctly, as this is an adaptation of the official Arduino Keyboard library, you would have to use the GNU Lesser General Public License v2.1 or later.

T-vK, 2020-11-30:
Yes, I think I can do that. At first I was hesitant because a lot of the code is based on this post https://github.com/nkolban/esp32-snippets/issues/230#issuecomment-473135679 and while I do have chegewara's permission to use his code under the MIT license, he wasn't sure how much of the code he actually wrote himself. 
But after revisiting his post, I think that my code is barely the same as his anymore after all my modifications. 

Now the only issue I see with adding a license is the fact that this library is also heavily based on the official USB keyboard library: https://github.com/arduino-libraries/Keyboard/blob/master/LICENSE
which is licensed under GPL v3.0.

So I don't really know what to do now. 

Maybe @chegewara would give us permission to use his code under the GPL v3.0 license?

This way I wouldn't have to deal with figuring out how to deal with multiple licenses.

anujdatar, 2020-11-30:
Aah, I see. Thanks for the update. I'll hold off submitting a pull request till we know more I guess. Thanks again.

chegewara, 2020-11-30:
I dont know about licensing, how it works and how you can change licence when your code is derived from library with Apache licence:
https://github.com/nkolban/esp32-snippets/blob/master/LICENSE

My part of code, which is HID i that library, can be considered MIT or Apache. 
As far as i know arduino-esp32 library is GPL 3, but that does not mean the library used in arduino has to be GPL, but app that is based on arduino-esp32 has to be GPL v3.

T-vK, 2020-11-30:
Well, if you only have a depency on another library, but don't include its source code in your repository, you can completely ignore its license. 
So all the licenses of the low-level BLE libraries that I'm merely depending on, can be fully ignored.

The two parts that can't be ignored are the usb keyboard library and the code that you wrote @chegewara, because these are the only parts that actually directly exist in this repo in terms of source code, at least to some degree.

Now the question is if I would have your blessing redistributing your code that I modified under the GPL v3 license as part of this library.

cwgstreet, 2022-11-04:
It would be really good to add an explicit licence to this useful project and close this long standing issue out as technically, your project is private domain without an explicit licence. 

I'm not a lawyer but reading comments above, it seems straightforward that when you combine code under permissive licence (MIT, Apache, BSD) with protective copyleft license such as GPL2 or 3, you  issue the derivative work under GPL3 licence. 

 Ref: [Wikipedia Licence Compatibility article](https://en.wikipedia.org/wiki/License_compatibility)

also see good discussion [here](https://www.quora.com/How-do-I-properly-credit-an-original-codes-developer-for-her-open-source-contribution)

NuggetronGames, 2026-11-15:
Reviving a dead thread to also request an explicit license. I'm developing a 3d printed foot pedal macro pad that uses this library. NimBLE had some compatibility issues. I'd really like to license this explicitly.

If licensing is still on the back burner, could i have your permission to distribute compiled arduino binaries using this library?