T-vK/ESP32-BLE-Keyboard #62: crashing android pixel 4 with "example code"

robomaniac, 2020-12-09:
this one is weird. while debugging to make it work on iOS, I decided to test on Android and use this example code
```
/**
 * This example turns the ESP32 into a Bluetooth LE keyboard that writes the words, presses Enter, presses a media key and then Ctrl+Alt+Delete
 */
#include <BleKeyboard.h>

BleKeyboard bleKeyboard;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
}

void loop() {
  if(bleKeyboard.isConnected()) {
    Serial.println("Sending 'Hello world'...");
    bleKeyboard.print("Hello world");

    delay(1000);

    Serial.println("Sending Enter key...");
    bleKeyboard.write(KEY_RETURN);

    delay(1000);

    Serial.println("Sending Play/Pause media key...");
    bleKeyboard.write(KEY_MEDIA_PLAY_PAUSE);

    delay(1000);

    Serial.println("Sending Ctrl+Alt+Delete...");
    bleKeyboard.press(KEY_LEFT_CTRL);
    bleKeyboard.press(KEY_LEFT_ALT);
    bleKeyboard.press(KEY_DELETE);
    delay(100);
    bleKeyboard.releaseAll();

  }
  Serial.println("Waiting 5 seconds...");
  delay(5000);
}
```

oh boy oh boy that my pixel 4 did not like this!!
it rebooted once
I turn it back on and open a notepad and try again. I was holding my esp32 in reset but pressing the reset button.
I let go the button...
I see the "sending hello world" and then phone reboot again
and see this message saying "phone corrupted, press power button to continue" and the phone is now stuck on google icon.
I hard reset it by pressing home and vol down for 7sec. I am stuck and phone does not boot up.

here are pictures of what happen
https://imgur.com/a/xZfSycK?s=fbm&fbclid=IwAR1kV2SpQZhmvbAJnjTuSBJ50f2cMFzsAsmfU1ysrHbrhFoyz6-dGuzP6x8

I would suggest you comment out the Ctrl+Alt+Del portion of the code 🤣 

[image]

[image]

T-vK, 2020-12-09:
Wow, I'm really sorry this happened. I will remove the Ctrl+Alt+Del from the example. I remember #10 reported a Mac OS problem where the default example would open iTunes and prompt to delete the first file.

A quick google search seems to indicate that the "Your device is corrupt" error message is a common problem with the Pixel 4.
Some people are suggesting trying to boot into safe mode. Others are suggesting to reflash the stock ROM which wouldn't be too difficult seeing you're already in fastboot mode, but this would delete all your data from the phone so I would only recommend it if you have a backup of all the data that is important to you.

T-vK, 2022-01-06:
This issue has been fixed by ScaredyCat in #134 and #135. It's on the master branch and will make it into the next beta release.

closed 2022-01-06 by T-vK