T-vK/ESP32-BLE-Keyboard #65: cant connect/pair

drunkscientist, 2020-12-24:
i feel like im missing something. device shows up as an available bluetooth device, but wont connect to anything. just says 'not set up' or  connection failed please try again. have tried with win10, linux, ios (original SE). what am i missing?

T-vK, 2020-12-29:
Since you have problems with multiple platforms, maybe while flashing something went wrong and the data got slightly corrupted or maybe its a power delivery issue. You may also try to use an older version of the library. And for the iPhone you might have to specify a shorter device name.

drunkscientist, 2020-12-29:
so i keep making minor changes and reuploading in the hope something will work. was able to get it to 'pair' in as far as my pc showed it requesting connection. after allowing it however it begins to rapidly cycle between connected/disconnected. serial monitor is all over the place: brownout detection, kernel 1 panic'd (cache locked but accessed or something), kernel 1 panic'd (watchdog timer something something) and sometimes its quiet. 3.4v direct from an atx supply to 3v3 pin (bypassing board regulator) for power, so its def not shoddy usb cable... waiting on delivery of another board to determine if i just have a bad board

drunkscientist, 2021-01-07:
so recieved another board and got the same errors. stripped it down to basics and got it working, so the issue is probably my implementation of interrupts

closed 2021-01-07 by drunkscientist

tonyxforce, 2023-03-15:
which version are you using?