T-vK/ESP32-BLE-Keyboard #66: combine with BLE-Mouse

street-grease-coder, 2020-12-29:
can this be done easily, or is this currently not feasible and mostly out of reach for arduino beginners?

street-grease-coder, 2020-12-29:
=> look at forks

T-vK, 2020-12-29:
At the moment it's not possible or feasible for a beginner. The only option for now is using the forks you mentioned that combine the two libraries into one.

street-grease-coder, 2020-12-29:
thanks for the quick and straight to point answer please feel free to either keep open to increase visibility for other users of your library or close

ServAlex, 2021-01-10:
@street-grease-coder I backported updates from keyboard and mouse repos to such "combo" fork
https://github.com/ServAlex/ESP32-BLE-Combo
I didn't check compatibility on macos or ios, but it seems working fine with windows, linux and android