T-vK/ESP32-BLE-Keyboard #68: Read Numlock/Capslock/Scrolllock state progress?

LeeNX, 2021-01-22:
Any progress or feedback on ```Read Numlock/Capslock/Scrolllock state```?  
  
I was looking at some examples of how to possible do this and see that ```KeyboardOutputCallbacks``` might be logging feedback ```special keys``` is that correct?

lalalandrus, 2021-01-26:
I too am trying to probe those states. It seems right now it just gets dumped into the debug messages and lost forever. I dont know how to send the data up the callback to the parent function. If anyone can help, greatly appreciated.

T-vK, 2021-01-26:
I haven't looked into it yet. I wasn't even aware that it was already implemented to the point that it would handle and log changes to the lock states.

@lalalandrus 
Is this where it gets logged?
https://github.com/T-vK/ESP32-BLE-Keyboard/blob/928c09320e3e6e344a63640c7de21d52a20c1d47/KeyboardOutputCallbacks.cpp#L16

The name `onWrite` seems confusing... maybe onWrite was referring to the computer writing the lock state back to the keyboard? I can't remember.

If that's the case, maybe we can add 3 variables for each of the lock states and save the states to them within the onWrite method.
Kind of like it's done with the `connected` state in `BleConnectionStatus`:

https://github.com/T-vK/ESP32-BLE-Keyboard/blob/928c09320e3e6e344a63640c7de21d52a20c1d47/BleConnectionStatus.cpp#L8

which can then be read from the `BleKeyboard` class:

https://github.com/T-vK/ESP32-BLE-Keyboard/blob/928c09320e3e6e344a63640c7de21d52a20c1d47/BleKeyboard.cpp#L110

In addition to that I think it would make sense to add another variable which simply holds a callback function that you have to pass to the constructor of `KeyboardOutputCallbacks` once in the beginning which is then called whenever a lock state changes.

T-vK, 2021-01-26:
@LeeNX @lalalandrus See #70

60999, 2022-05-08:
Excuse me, what's the progress now? Has the problem been solved?