T-vK/ESP32-BLE-Keyboard #71: Cant pair to ESP32

kkuez, 2021-02-06:
Unfortunately I cannot state any stacktrace or similiar.

- Example sketch
- PlatformIO/ Arduino IDE does not matter
-  ESP gets detected
- Trying to pair
- Android phones state "Couldnt pair due to incorrect key/password entry"
- ESP starts working as if device is connected (Enters Keys af)

I changed psecurity Authmode to ESP No Bond in BleKeyboard.cpp and rebuild
-> Nothing changed

Tried with 0.2.2 as well as 0.2.1
I remember it working fine half a year ago

T-vK, 2021-02-06:
Try disconnecting, unpairing and removing the ESP form your Android device, turn Bluetooth off (on your phone), wait a few seconds, then turn it on again and try to pair and connect again.

kkuez, 2021-02-06:
It helped. Thanks!
Is this a known issue which might be solved in the future?
Regards

T-vK, 2021-02-06:
Yes, it's a known issue and I hope we can fix it in the future. It's quite possible that one of the open Pull Requests already fixes this issue, but it's a complicated situation as some of the Pull Requests break compatibility with some Apple devices.

Fusseldieb, 2021-05-05:
Just tried out this library!

They keypresses work wonderfully, but it almost never wants to connect. I need to re-pair it everytime and most of the time it just refuses to connect...

Often it spams the message

`[E][BLECharacteristic.cpp:528] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error`

in the log

bauerbyter, 2021-11-17:
Have the same problem. Does someone have a solution?

bauerbyter, 2021-11-18:
Version 0.3.0. is now working!

Fusseldieb, 2021-11-18:
> Version 0.3.0. is now working!

Has the owner updated the repo since then? If yes, I will absolutely give it another try :D

bauerbyter, 2021-11-19:
In my case the latest beta didnt work. But 0.3.0 did.