T-vK/ESP32-BLE-Keyboard #72: Connection indicator + What am I connected to ?

Ukhauk, 2021-02-08:
My question is does the ESP32 know what host name its connected to ?

Was thinking it might be useful to serial print the host name once on connection 

just in case several devices are trying to connect ?

my board is showing bluetooth connected but I don't know to which host

I have added the following to indicate if the ESP32 is connected (Wemos D1 R32 board) hope its useful to someone
//const int ledPin = 2;

//setup-
pinMode (ledPin, OUTPUT);

//loop-
if (bleKeyboard.isConnected()) {
      // if the current state is HIGH then the button went from off to on:
      digitalWrite (ledPin, HIGH);  // turn on the LED
    } else {
      // if the current state is LOW then the button went from on to off:
      digitalWrite (ledPin, LOW);  // turn off the LED
    }

T-vK, 2021-02-08:
Good point. That would indeed be useful. Maybe I'll add support for that when I find some time to work on this again.

Ukhauk, 2021-02-10:
Thanks for reply and sharing your code. Great project for ESP32 just need to work on the hardware now ( keyboard / enclosure ) like the idea of cherry keys

T-vK, 2021-02-12:
I recently saw an inspiring project here https://www.reddit.com/r/MechanicalKeyboards/comments/jyqozn/just_finished_my_diy_programable_bluetooth_macro/
It's not using cherry keys, but still, it looks pretty good.

T-vK, 2022-01-20:
I haven't looked into it yet, so I'm not sure.

Hannah-GBS, 2022-07-18:
Would also be interested in this. I was thinking of having a small e-ink display on my device that would display what it's connected to (along with some other data).

anandameyer, 2022-12-12:
I have tried to get a name of the connected device and found out there is no way we can get a connected device name without scanning/discovery nearby device since the mechanism to connect doesn't need name in the first place. 

Another problem is devices like notebook, personal computer, or smart phone advertise their name in (can i say legacy?) bluetooth advertisement, not in BLE advertisement.

Only connected BLE address I can get on my testing.

So the workaround maybe not to display connected device, but limit what device can connect to by changing BLE MAC Address upon initialization.