T-vK/ESP32-BLE-Keyboard #73: Trying to add another BLE service besides the keyboard service

renederidder, 2021-02-09:
I have an ESP32 that works very well acting as a BLE keyboard for an app running on an Android tablet. Now I need this ESP32 to act simultaneously as a BLE server with a custom service to another ESP32 acting as client.

So after the bleKeyboard.begin(); I can add pServer = BLEDevice::createServer(); to create the server for that custom service, and the keyboard still works fine.

But as soon as I add the line pService2 = pServer2->createService(SERVICE_UUID); the keyboard stops sending keystrokes.

Why would this added line break the BLE keyboard service?

T-vK, 2021-02-11:
Unfortunately I don't know enough about the BLE libraries to answer your question. Maybe the people who wrote those libraries can help. The BLE libraries are only OOP wrappers around the [official function based API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/index.html).

Matth66, 2022-12-27:
Someone got idea how to fix it?