T-vK/ESP32-BLE-Keyboard #74: Send keyboard power key

ChMG, 2021-02-09:
Hi,

I am trying to emulate a remote control of a media reciever.
For all buttons at the remote controll I found the corresponding keybord keys.

Only the power button makes problems.
From the specs I know the button uses the HID code 0x66 (102).

I have added 136 to send a non-printing key.
So the command is `bleKeyboard.write(238)`.

If I trigger this command nothing happens.

Does anyone have any idea why this command doesn't work?

ChMG, 2021-02-11:
Hi,

in the meantime I´m totaly confused.
We found a way to trace the bluetooth communication.

At this trace I can see the correct HID code (green marked), but the device dosn´t power off :-(

The only different is the handle code (yellow marked), but I don´t know if this is relevant:
- original remote: 0x0012
- esp32: 0x0041

trace original remote:
[image]

trace esp32:
[image]

I have no idea what the problem is.

T-vK, 2021-02-11:
Unfortunately I don't know enough about Bluetooth to really answer your question.
I would try modifying the report descriptor. E.g. replace 
`  USAGE(1),           0xB7,          //   USAGE (Stop)                ; bit 2: 4`
with 
`  USAGE(1),           0x30,          //   USAGE (Power)               ; bit 2: 4`

https://github.com/T-vK/ESP32-BLE-Keyboard/blob/928c09320e3e6e344a63640c7de21d52a20c1d47/BleKeyboard.cpp#L74

And replace 
`const MediaKeyReport KEY_MEDIA_STOP = {4, 0};`
with 
`const MediaKeyReport KEY_MEDIA_POWER = {4, 0};`

https://github.com/T-vK/ESP32-BLE-Keyboard/blob/928c09320e3e6e344a63640c7de21d52a20c1d47/BleKeyboard.h#L65

And then try to use it like this `bleKeyboard.write(KEY_MEDIA_POWER);`.

ChMG, 2021-02-11:
Hi, 
this was a very good hint.
Now I can set the media reciever to standby and wake them up again.

Thank you :-)

T-vK, 2021-02-12:
No problem, glad to hear it's working. :)

ChMG, 2021-02-12:
I found another strange behavior.
If I send `bleKeyboard.press(KEY_MEDIA_POWER)` and `bleKeyboard.releaseAll()` the media reciever crashes and perform a restart.

The combination `bleKeyboard.press(KEY_MEDIA_POWER)` and `bleKeyboard.release(KEY_MEDIA_POWER)` works as aspected.

OK, there is issue #42 for media keys and releaseAll, so I think we can close this issue.

closed 2021-02-12 by ChMG