T-vK/ESP32-BLE-Keyboard #75: Will the use of bleKeyboard work as a BT Remote Control  ?

TheStigh, 2021-02-16:
Hi, kind of off-topic but do you know if this solution would work as a BT Remote Control for my Satellite Receiver? I've not found any good articles yet how to simulate a BT Remote Control. Now, sat receivers and a lot of new tv's are sold with BT controllers rather than IR...

In advance, thanx for any help/input.

T-vK, 2021-02-17:
Probably not, but maybe. If you can remote control it with a normal bluetooth keyboard, I would say there is a chance that you could remote control it with this library.
I think odds are you're better off trying to remote control it via Wifi or IR. Maybe using a library like this https://github.com/crankyoldgit/IRremoteESP8266
Even if the satellite receive is only shipped with a BT remote, it might still have an IR receiver or a Wifi interface.

TheStigh, 2021-02-17:
@T-vK thanx, it does have IR and I've already built a project based on ESP32 where I have both an IR Rec and IR Trans where I learn the codes from the original remotes and make scripts to swap channels etc from Home Assistant (and by voice via Google Assistant). The 'issue' is the LoS. With IR I need to place my box, which also need power, somewhere it can see the receiver/tv. With BT I can simply hide it behind the receiver etc.

As soon as I return from travel I will try to pair a BT Keyboard to my sat receiver (an OEM manufactured by Technicolor) from Canal Digital. As you say, if the keyboard works, your code should also work :) So exited to test this!

Thanks for the feedback, I will update you on the progress.

T-vK, 2021-02-17:
You could put the ESP or at least the IR transmitter LED right in front of the IR sensor of your receiver and then use your phone to send commands to the ESP via Wifi.
Since you use Home Assistant, maybe flashing your ESP with Tasmota would be an option for you. This way you should be able to integrate it with Home Assistant with relative ease.

TheStigh, 2021-02-17:
I use MQTT to communicate with HA, and IR works great, both learning and sending commands. It is the Line of Sight I want to go around, that means BT. But I will test with a BT Keyboard and see if it works on the sat receiver.

T-vK, 2021-02-17:
I understand that, but line of sight wouldn't be an issue if you placed the ESP with the IR transmitter close to the satellite receiver's IR sensor and then remote controlled the ESP over Wifi because in that case your range would only be limited by your Wifi.

TheStigh, 2021-02-17:
serviceUUID: 00001812-0000-1000-8000-00805f9b34fb - Isn't this classified as a HID?

TheStigh, 2021-02-17:
UPDATE: 

I used the example from **chegewara** (https://github.com/chegewara/esp32-hid-keyboard-client) and was able to connect to the remote :) I could then see the numbers from the keypad on the console output. This means this is possibly the solution, use BleKeyboard.

Though, I could not see any output when pressing Power, any Arrow-keys, OK or Volume buttons. Could this mean the keys for these buttons are not propers defined in the 'const char KEYS[]' ?

Reason for not testing this directly on the sat receiver now is that I'm on travel, but brought the remote to have something to work on while being bored in a hotel room... :)

I can add that the sat receiver are based on Android.

TheStigh, 2021-03-02:
To finalize this - the answer to my own question are, Yes.

closed 2021-03-02 by TheStigh

T-vK, 2021-03-02:
So did you manage to properly remote control your receiver using your esp?

TheStigh, 2021-03-02:
> 
> 
> So did you manage to properly remote control your receiver using your esp?

Indeed!

Though, started to add this library to the rest of my code, and it seems it eats up memory totally and my ESP panic. I've seen the posts about to much bytes (20000) and tried to reduce to 4096, still panic's on bleKeyboard.begin().

T-vK, 2021-03-02:
Yeah memory usage is a big problem at the moment. Maybe you can use PR #54 .

TheStigh, 2021-03-02:
> 
> 
> Yeah memory usage is a big problem at the moment. Maybe you can use PR #54 .

Hummm... seems I do have enough memory, 

`14:05:54.664 -> Core 1 was running in ISR context:`

Interrupt issue?
I've tried to add 
`#define INTERRUPT_ATTR` and `IRAM_ATTR`

...but doesn't seem to help...

T-vK, 2021-03-02:
I have no idea what the error means unfortunately. But it doesn't sound like the typical error messages that I used to get when my ESP ran out of memory.