T-vK/ESP32-BLE-Keyboard #77: How to read/detect pressed key on the actual keyboard?

radeknemec22, 2021-03-03:
Hi, first of all thank you very much for your project, I really like it, good job! 
I would like to ask you, if there is a way how to detect pressed key on the keyboard - e.g. my program is running and I press ESC key on the actual keyboard a I'd like my ESP to detect it, so I am basically looking for a read function. I can't figure it out. Thank you!

T-vK, 2021-03-03:
No that's not possible unless you have the keyboard attached directly to the ESP using a USB host shield.

mccwong, 2021-03-29:
@radeknemec22 
Hi radeknemec22, I personally have not tried this library, but it seems this library works with a PS2 (or some USB) keyboard connected to a ESP32 via 2 GPIO pins (plus VCC and Gnd).   
The authors actually documented a very interesting and major project using this library and a few others, for emulating a Nintendo on a ESP32.

https://github.com/nathalis/ESP32-USB-Keyboard

https://github.com/nathalis/NesCat/blob/main/Libraries/Ps2KbdBle/Ps2KbdBle.ino
https://github.com/nathalis/NesCat

T-vK, 2021-03-30:
I think radeknemec22 was asking for a way to read the keystrokes of a second keyboard that is connected directly to the computer. USB or Bluetooth keyboards cannot read each others keystrokes with the exception of the Capslock, Numlock and Scrolllock keys.

`ESP32-USB-Keyboard` is about plugging a keyboard directly into the ESP, similar to how you can do it with a USB host shield.
I think the name is a bit misleading though. I might be wrong, but I think it's limited to keyboards that actually support the PS/2 protocol. Some keyboards allow using the PS/2 protocol on the data lines of the USB connector in order to allow for USB-to-PS/2 adapters. Since this is not spec-compliant and demand for PS/2 has basically vanished, I would think that newer keyboards are less likely to have that feature. 

The ESP-32S2 will, according to Espressif, support USB OTG eventually which would make it possible to actually connect any kind of USB device to the it (without a USB host shield). But it may take another year or more until they release the necessary software and documentation.

mccwong, 2021-03-30:
@T-vK , I believe you are correct in both of the points you have stated in your reply.   I stand corrected and as the info in my earlier post became irrelevant I think it is best the post be deleted lest it caused confusion in the future.