T-vK/ESP32-BLE-Keyboard #81: Problem sending F13 to F24 keys.

rp1231, 2021-03-25:
Hi,
I would like to report a problem sending the keys F13 to F24.
I am able to send all the keys otherwise but these keys are not going through.

I have just tested the key F13 and F6 specifically but have mentioned the 13 to 24 range as those are not the standard keys available on the keyboard.

Also posting a link to a similar problem faced by other people using the standard keyboard.h library.
https://forum.arduino.cc/index.php?topic=324554.0
In the post it is mentioned that the Arduino keyboard code falsely translates the keycodes that are sent via Keyboard.press()
It also has a few solutions implemented by people for the usb keyboard.h library.

Maybe that has something to do with this?

T-vK, 2021-03-25:
Your link always redirects me to the home page of https://forum.arduino.cc

rp1231, 2021-03-25:
Strange.....
It does not do that for me.....

rp1231, 2021-03-25:
The problem has been resolved.
I got help from Orotavia and MetalMouse from the Deej forums.

The problem was that default descriptor had cut off those last function keys in ble_keyboard.cpp lines 56 and 59.

Changing the values
from:
LOGICAL_MAXIMUM(1), 0x65 and USAGE_MAXIMUM(1),   0x65,
to:
LOGICAL_MAXIMUM(1), 0x73 and USAGE_MAXIMUM(1),   0x73,

solved the problem.

wjcarpenter, 2021-03-25:
OP's link worked for me.

T-vK, 2021-03-25:
Weird, I tried Firefox and Chromium - same behavior. Anyway, glad to hear the issue is resolved now. :)
I'm sure it will help other people in the future.

rp1231, 2021-03-25:
Maybe this can be changed in the code so that others don't go through it ?

T-vK, 2021-03-26:
Yes, we can do that if we can verify that it works on Windows, Linux, Mac OS, Android and iOS. I can test for Android and Linux.

rp1231, 2021-03-26:
I have tested on windows

closed 2021-03-27 by rp1231