T-vK/ESP32-BLE-Keyboard #83: Shift key is pressed without code

DaisukeMinagawa, 2021-03-30:
I cannot understand why the shift key is pressed when "Ctrl + N" and "Ctrl + P".

```
#include <M5StickCPlus.h>
#include <BleKeyboard.h>
#define PWR_BTN_SHORT_PRESS 2

BleKeyboard bleKeyboard("ZOOM", "...");

enum ConfSystem {
  vaps
};
//Switch mic, video, share mode
String switch_mode = "mic";
int n_conf = 2;
bool on_air;
 
double chargingBattery = 4.9;
double maxBattery = 4.08;
double minBattery = 3.5;
double defBattery = maxBattery - minBattery;
double currentBattery;

int screen = 0;

float accX = 0;
float accY = 0;
float accZ = 0;

void show_need_connect() {
  M5.Lcd.fillScreen(TFT_BLACK);
  M5.Lcd.setTextColor(TFT_WHITE, TFT_BLACK);
  M5.Lcd.setTextSize(3);
  M5.Lcd.setTextDatum(MC_DATUM);
  M5.Lcd.drawString("Plz Connect", 120, 70);
}

void show_onair() {
  M5.Lcd.fillScreen(TFT_BLUE);
  M5.Lcd.setTextColor(TFT_WHITE, TFT_BLUE);
  M5.Lcd.setTextSize(5);
  M5.Lcd.setTextDatum(MC_DATUM);
  M5.Lcd.drawString("Execute!", 120, 70);

  M5.Lcd.setTextSize(2);
  M5.Lcd.setTextDatum(BR_DATUM);
  if (switch_mode == "mic") {
    M5.Lcd.setTextSize(3);
    M5.Lcd.drawString("Mic", 240, 130);
  }
  else if (switch_mode == "video") {
    M5.Lcd.setTextSize(3);
    M5.Lcd.drawString("Video", 240, 130);
  } else {
    M5.Lcd.setTextSize(3);
    M5.Lcd.drawString("Share", 240, 130);
  }
}

void show_inmute() {
  M5.Lcd.fillScreen(TFT_DARKGREY);
  M5.Lcd.setTextColor(TFT_WHITE, TFT_DARKGREY);
  M5.Lcd.setTextSize(5);
  M5.Lcd.setTextDatum(MC_DATUM);
  M5.Lcd.drawString("Execute", 120, 70);

  M5.Lcd.setTextSize(2);
  M5.Lcd.setTextDatum(BR_DATUM);
  if (switch_mode == "mic") {
    M5.Lcd.setTextSize(3);
    M5.Lcd.drawString("Mic", 240, 130);
  }
  else if (switch_mode == "video") {
    M5.Lcd.setTextSize(3);
    M5.Lcd.drawString("Video", 240, 130);
  }
  else if (switch_mode == "share") {
    M5.Lcd.setTextSize(3);
    M5.Lcd.drawString("Share", 240, 130);
  }
}

void send_mute_mic() {
  bleKeyboard.press(KEY_LEFT_GUI);
  bleKeyboard.press(KEY_LEFT_SHIFT);
  bleKeyboard.press('a');
  delay(100);
  bleKeyboard.releaseAll();
}

void send_off_video() {
  bleKeyboard.press(KEY_LEFT_GUI);
  bleKeyboard.press(KEY_LEFT_SHIFT);
  bleKeyboard.press('v');
  delay(100);
  bleKeyboard.releaseAll();
}

void send_off_share() {
  bleKeyboard.press(KEY_LEFT_GUI);
  bleKeyboard.press(KEY_LEFT_SHIFT);
  bleKeyboard.press('s');
  delay(100);
  bleKeyboard.releaseAll();
}

void update_display() {
  if (on_air) {
    M5.Beep.tone(4000);
    delay(10);
    M5.Beep.mute();
    show_onair();
  } else {
    M5.Beep.tone(2000);
    delay(10);
    M5.Beep.mute();
    show_inmute();
  }
}

void setup() {
  M5.begin();
  bleKeyboard.begin();
  M5.IMU.Init();

  M5.Lcd.setRotation(3); // BtnA is left side to LCD
  M5.Beep.setBeep(4000, 100);
  //Make display daker, it is recommended more than 7
  M5.Axp.ScreenBreath(8);
  setCpuFrequencyMhz(80);

  on_air = true;
  update_display();
}

void loop() {
  M5.update();

  int axpButton = M5.Axp.GetBtnPress();
  if (axpButton == PWR_BTN_SHORT_PRESS) {
    if ( screen == 1 ) {
      M5.Axp.ScreenBreath(8);
      screen = 0;
    } else {
      M5.Axp.ScreenBreath(0);
      screen = 1;
    }
  }

  if (M5.BtnA.wasPressed()) {
    if (bleKeyboard.isConnected()) {
      on_air = !on_air;
      if (switch_mode == "mic") {
        M5.Beep.tone(5000);
        delay(10);
        M5.Beep.mute();
        send_mute_mic();
      }
      else if (switch_mode == "video") {
        M5.Beep.tone(2000);
        delay(10);
        M5.Beep.mute();
        send_off_video();
      }
      else if (switch_mode == "share") {
        M5.Beep.tone(1000);
        delay(10);
        M5.Beep.mute();
        send_off_share();
      }

      update_display();
    } else {
      M5.Beep.tone(4000);
      delay(10);
      M5.Beep.mute();
      show_need_connect();
      delay(3000);
    }
    update_display();
  }

  if (M5.BtnB.wasPressed()) {
    if (switch_mode == "mic") {
      switch_mode = "video";
    }
    else if (switch_mode == "video") {
      switch_mode = "share";
    }
    else if (switch_mode == "share") {
      switch_mode = "mic";
    }

    update_display();
  }
  M5.Beep.update();
  //When battery is low
  if (M5.Axp.GetWarningLeve() == 1) {
    M5.Lcd.setTextSize(3);
    M5.Lcd.drawString("LowBattery", 0, 0);
    M5.Beep.beep();
    delay(10);
  } else {
    //Get Voltage of internal battery
    M5.Lcd.setTextSize(3);
    currentBattery = M5.Axp.GetAPSVoltage();
    double battery = (currentBattery - minBattery) / defBattery * 100;
    if (battery > 100) {
      M5.Lcd.drawString("100%       ", 0, 0);
      //M5.Lcd.drawString("100%       ",0,0);
    } else if (battery < 0 ) {
      M5.Lcd.drawString("  0%", 0, 0);
      M5.Beep.beep();
    } else if (battery < 10) {
      M5.Lcd.drawString(String((int)battery) + "  %", 0, 0);
    } else {
      M5.Lcd.drawString(String((int)battery) + " %", 0, 0);
    }
    delay(100);
  }

  /*Control to go next 25 and go previous 25   <-The shift key is pressed inside this code without bleKeyboard.press(KEY_LEFT_SHIFT); or something....*/

  M5.IMU.getAccelData(&accX, &accY, &accZ);
  if ((accX * 1000) < -400) {
    Serial.println("going next 25...");
    M5.Beep.tone(3000);
    delay(10);
    M5.Beep.mute();
    bleKeyboard.write(KEY_LEFT_CTRL);
    bleKeyboard.write('P');
    delay(200);
    bleKeyboard.releaseAll();
  }
  if ((accX * 1000) > 400) {
    Serial.println("going prevuouse 25...");
    M5.Beep.tone(3000);
    delay(10);
    M5.Beep.mute();
    bleKeyboard.write(KEY_LEFT_CTRL);
    bleKeyboard.write('N');
    delay(200);
    bleKeyboard.releaseAll();
  }
}
```

T-vK, 2021-03-30:
Can you provide a minimal code example that reproduces this behavior? E.g. without the M5 specific code?

DaisukeMinagawa, 2021-04-03:
Here you are. Even I haven't written the shift key code in my code. When the device(ESP32-WROOM-32D) pushed the button, the shift key also pushed.

```
#include <BleKeyboard.h>

BleKeyboard bleKeyboard("sample"); 

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();

  pinMode(32,INPUT);    //GPIO32
  pinMode(33,INPUT);    //GPIO33
}

void loop() {
  if(bleKeyboard.isConnected()) {    
    if(digitalRead(32) == HIGH){    //32
    bleKeyboard.write(KEY_LEFT_CTRL);
    bleKeyboard.write('N');    
      delay(200);
    }
    if(digitalRead(33) == HIGH){    //33
    bleKeyboard.write('P');
      delay(200);
    }
  }
}
```

DustinWatts, 2021-04-03:
Try this:

```
bleKeyboard.press(KEY_LEFT_CTRL);
bleKeyboard.write('n'); 
bleKeyboard.releaseAll(); 
```

Note that 'N' is different from 'n'. To get a capital 'n' (N), shift is needed, so shift is pressed ;)

DaisukeMinagawa, 2021-04-03:
@DustinWatts 
Thanks. The issue is solveld!