T-vK/ESP32-BLE-Keyboard #84: Espressif32@3.1.0 stopped advertising automatically on disconnect

aovestdipaperino, 2021-04-21:
I don't know if this change is intentional or not, but after 3.1.0 the advertising is not restarted when the device is disconnected. These are the repro steps on iOS:

1. create a simple BleKeyboard
2. connect to it on iOS (same thing observed on Windows)
3. forget the device on iOS
4. try to reconnect, the pairing will timeout.

I verified that restarting the advertising after disconnect fixes the issue.
I can generate a PR, these are the options:

a. store the pAdvertising instance in BLEConnectionStatus and restart it on disconnect (preferable as it doesn't break the previous code counting on this)
b. expose a startAdvertising method in BLEKeyboard for the users to do whatever they feel right

T-vK, 2021-04-21:
This is not intentional. It's a bug many have noticed. For example in #56 . In PR #53 rousir has exposed startAdvertising like you mention in `b`, but I feel like that shouldn't be something that the user has to deal with. And it would of course break backwards-compatibility. In my opinion this should be done automatically.
So what you're proposing in `a` sounds like the way to go. 

I'd gladly accept a PR for this if we can verify that it works on Android, iOS, Windows, Mac OS and Linux. 
I can only test Android and Linux. And I assume you can test on Windows and iOS. We can probably find someone who's willing to test it on Mac OS.

aovestdipaperino, 2021-04-21:
I can test Windows, MacOS and iOS. I will report back.

aovestdipaperino, 2021-04-21:
I've put a simple ESP32 to reproduce the problem easily here: https://gist.github.com/aovestdipaperino/921413d6b3de17670fa8a8f39d1a4f5d

Tested the patch on iOS. 
Testing on Windows and MacOS next.

aovestdipaperino, 2021-04-21:
PR here: https://github.com/T-vK/ESP32-BLE-Keyboard/pull/85

aovestdipaperino, 2021-04-21:
Updated PR: https://github.com/T-vK/ESP32-BLE-Keyboard/pull/86