T-vK/ESP32-BLE-Keyboard #88: Sending long texts via .print()

paz0ok, 2021-04-28:
Hi everyone,

I'm currently building a Bluetooth Macropad with an ESP32. I use the Arduino IDE to program everything. One profile I use is for programing. For this, I want to send text.

Everything is working to an extend, but when I am trying to print larger Stings (longer than 10 letters), only a part of the text is send when I normaly press the button.

Can anyone help me here?

`#include <BleKeyboard.h>
BleKeyboard bleKeyboard;

#define button1 32

void setup() {
  pinMode (button1,INPUT_PULLUP);
  Serial.begin(9600);
  bleKeyboard.begin();
}

void loop() {
  if(bleKeyboard.isConnected()) {
    if (digitalRead(button1) == LOW){
      bleKeyboard.print("Print some text to the PC");  
      delay(200);
    }
  }
}`

DustinWatts, 2021-04-28:
You could iterate through your string of text and add a small delay between each character. Something like:

```
const char string[] = "Print some text to the PC";

for(int i=0; i < strlen(string); i++ ) {
  char c = string[i];
  bleKeyboard.print(c);
  delay(10); // maybe increase this a bit
}

I had success using this method printing very large strings, but I can't find the exact code I used anymore.

paz0ok, 2021-05-09:
@DustinWatts Thanks for your help! This works perfectly 👍

closed 2021-05-09 by paz0ok