T-vK/ESP32-BLE-Keyboard #89: Hello zorld: Keymapping issue

fmatray, 2021-05-03:
Hello,
I tried the exemple on my mac. After some BLE modifications, I can pair the device.

One funny thing, I get "Hello **z**orld" because I use a french on my computer.
Is there a way to implement different mappings with this library ?

Thanks

T-vK, 2021-05-03:
The keyboard layout is in BleKeyboard.cpp in the _asciimap variable. At the moment it's hardcoded to US. If you wan't a different one you either have to write a translator function or change your BleKeyboard.cpp file. 
You might also be able to tell your mac to use a US layout for the ESP keyboard if you only need this on one computer.

wjcarpenter, 2021-05-03:
The BLE HID spec is largely a complete adoption of the USB HID spec. The USB HID spec does a peculiar thing for keyboard physical layouts.  It puts the burden on the computer side to know about layouts. So, when you type a "Z" on a QZERTY keyboard, it actually sends a "W", and the computer knows that that should be a "Z" because you are using a QZERTY keyboard. The rationale for this is that it saves the peripheral manufacturers from having to mess around with firmware for different layouts. 

There is an option for the keyboard to tell the computer what its physical layout is (see section 15.18 of this spec: https://www.usb.org/sites/default/files/hut1_22.pdf), so you might be able to modify BleKeyboard to send such a report to the computer.

(How anybody could think this quirk of the USB HID spec is good engineering is beyond me. Maybe some powerful keyboard makers refused to do it the logically reasonable way, so USB codified the practice they would accept.)

fmatray, 2021-05-03:
Thanks for your answers.

Well, as I use my regular keyboard in french and the ESP keyboard is in addition for some shortcuts, I can't switch the mapping on computer side all the time.

T-vK, 2021-05-04:
Just compare a picture of a US keyboard layout with yours and you should be able to tell which entries in the asciimap represent your extra letters.