T-vK/ESP32-BLE-Keyboard #91: Sending ASCII Hex or Dec instead of char

paz0ok, 2021-05-09:
Hi everyone,

I have problems with sending Chars or Strings becaus I use ISO-DE Layout.
So when I'm trying to send "[" it does not work. I was wondering if I can send Hex oder Dec ASCII Chars instead of typing them as bleKeyboard.print("[").

I could not find a solution online.
Can someone help me?

T-vK, 2021-05-09:
Exactly the same way as you would do it for the Arduino USB keyboard library.........
https://forum.arduino.cc/t/hid-device-cheap-usb-rubber-ducky-german-layout-problem/419974/5

I suppose Right Alt key + 9

``` C++
bleKeyboard.press(KEY_RIGHT_ALT);
bleKeyboard.print("9");
bleKeyboard.release(KEY_RIGHT_ALT);
```

iron2serzh, 2021-05-10:
bleKeyboard.press(91); // 91 = key "[" in ASCII Table http://www.asciitable.com/

paz0ok, 2021-05-22:
Thanks for your help.
@T-vK Your solution Worked for me 😃

closed 2021-05-22 by paz0ok