T-vK/ESP32-BLE-Keyboard #94: feature request - support left and right windows keys

roblatour, 2021-06-01:
Thanks for your work on this library.

It would be great if support for the left and right windows keys could be added.

I tried a combination of the control and escape keys, and while they are somewhat similar they don't offer the same end result as s the Windows keys.

T-vK, 2021-06-02:
ESP32-BLE-Keyboard supports all the keys that the standard Arduino USB Keyboard library supports. This includes both Windows keys.

roblatour, 2021-06-02:
When I look at the file BleKeyboard.h I don't see them listed, how/where are they defined?

T-vK, 2021-06-02:
They are defined in `BleKeyboard.h`. Check https://arduino.stackexchange.com/questions/23792/press-the-windows-key-using-keyboard-press

roblatour, 2021-06-02:
Thank you very much for the clarification, I didn't associate the Windows key with the 'GUI key'.

closed 2021-06-02 by roblatour