T-vK/ESP32-BLE-Keyboard #95: Two BleKeyboard instances in same sketch?

jason-kong, 2021-06-08:
Hi There, I've been trying to figure this out for a couple of days now, but is it possible to have two BleKeyboard instances in the same Arduino IDE sketch? (I plan on pairing each instance with two different computers/phones and then programatically switch between the two in various scenarios)

From the code that I can see, most of the codepaths that each instance has seems fairly isolated but when I run the sketch, only one of the instances advertises, I can see both taskServer methods go in but at some point only one instance seems to keep running. Any tips on whether this is possible or not? And if not, what's the spot of contention ?? Thanks!

T-vK, 2021-06-08:
At the moment this is not possible. It's also not possible to combine `ESP32_BLE_Mouse`, `ESP32_BLE_Keyboard` etc with each other. I'm not sure if this is a limitation of the SDK, the BLE wrapper classes around the SDK or just bad programming from my side on this library. `ESP32_BLE_Mouse` and `ESP32_BLE_Keyboard` were essentially just a quick and dirty weekend project that unexpectedly became very popular. 

Someone has created a new library that combines `ESP32_BLE_Mouse` and `ESP32_BLE_Keyboard` into one library. Maybe you could use that as a starting point to make it work with two instances of `ESP32_BLE_Keyboard`. 

`https://github.com/blackketter/ESP32-BLE-Combo

jason-kong, 2021-06-08:
@T-vK totally understand; I was looking at the BLE combo project just before I posted yesterday unfortunately I think it reports one device with multiple functions, where as I want to take one device and advertise it twice so that two separate connections can be attached to it, even if the keyboard can only talk to one at a time. I will keep digging, just nothing obvious jumps out at me from the code inside your project. If I manage to come up with something I will report back. Thanks

jason-kong, 2021-06-30:
I was able to achieve what I wanted to do in principle;  one BLE Keyboard, paired and bonded with two computers; and added the ability to switch between them..

Basically, in the onConnect callback, I continue to advertise
```
void BleConnectionStatus::onConnect(BLEServer* pServer)
{
...
pServer->startAdvertising();
}
```
This allows a second device (and so on) to connect to the keyboard.

Following that, I found that whatever was being typed was being sent to both computers due to how the `notify` logic works. So I added the ability to set the current active device (either connection 0 or 1), and placed a flag that would only `notify` input over to the active device; 

Because of this kind of implementation, switching between devices is super fast, since they are already pre-bonded; There were  some little quirks that I was also able to solve, and there is the slow initial BLE connection time on boot, but I'm not sure there's much that can be done about that

tahunus, 2021-12-04:
Hi @jason-kong 
I'm trying to do a similar thing: one BLE Keyboard paired with two clients. In my case, one client is an iOS device for display and  text-to-speech functions (the production unit is ready but being documented, the prototype is here: [https://www.hackster.io/pedro-martin/bluetooth-rpm-letterboard-7f2ff4](url)  . The second client is a haptic feedback unit (a  nRF52840 with a haptic motor controller and vibrating disc). This way, the letter touched on the BLE Keyboard (it's a capacitive touch sensor) sends the letter to the iOS device for display and speech and to the nRF52840 for haptic feedback.

I looked into BLECharacteristic.cpp and did not find the function `BleConnectionStatus::onConnect(BLEServer* pServer)`

Where did you make the modification to continue to advertise and allow a second connection?

Thanks!