T-vK/ESP32-BLE-Keyboard #96: Bluetooth connection status keeps switching between connected and disconnected

kotobuki, 2021-06-23:
Bluetooth connection status returned by `isConnected()` keeps switching between `true` and `false` when not connected to a PC. I confirmed with the following environment. This might be an issue in lower layers, but I want to report here to make sure.

- Library: ESP32-BLE-Keyboard v0.2.3
- Board: ESP32 Arduino Release 1.0.6 based on ESP-IDF v3.3.5
- ESP32: M5Stack M5GO
- PC: macOS Big Sur and Windows 10

```c++
#include <BleKeyboard.h>

BleKeyboard bleKeyboard;

void setup() {
  Serial.begin(115200);
  bleKeyboard.begin();
}

void loop() {
  if (bleKeyboard.isConnected()) {
    Serial.println("connected");
  } else {
    Serial.println("not connected");
  }
  delay(500);
}
```

T-vK, 2021-06-23:
Interesting, thanks for reporting. I think it is a problem with [BLEServer](
https://github.com/espressif/arduino-esp32/blob/21947ebe76cf6380fe06aa885390b86cc477709a/libraries/BLE/src/BLEServer.cpp#L306) used [here](https://github.com/T-vK/ESP32-BLE-Keyboard/blob/361806ab352c4910073504883e799d7ac9adf9d2/BleKeyboard.cpp#L123), which is responsible for calling [onConnect](https://github.com/T-vK/ESP32-BLE-Keyboard/blob/361806ab352c4910073504883e799d7ac9adf9d2/BleConnectionStatus.cpp#L6) and [onDisconnect](https://github.com/T-vK/ESP32-BLE-Keyboard/blob/361806ab352c4910073504883e799d7ac9adf9d2/BleConnectionStatus.cpp#L16).

kotobuki, 2021-06-23:
@T-vK 
Thank you very much for your quick reply.

I just found the cause of the problem after testing with several devices. I have connected the device to my iPhone Xs (iOS 14.6). After finished testing, I disconnected but forgot to remove it. It seems that the iPhone tries to connect to the device. Once I removed the device on the iPhone, I can't reproduce this behavior.

I'm sorry for the false alarm. I'll mention this as a troubleshooting section of my project's document.

closed 2021-07-04 by kotobuki