T-vK/ESP32-BLE-Keyboard #97: Device disconnects then reconnects on raspberry pi 3b+ - but HID input not accepted

drdobbins, 2021-06-28:
After booting the pi and powering on the ESP32 HID device, everything connects and functions properly. 

When taking the ESP32 out of range, and bringing it back it _sometimes_ will not function, despite all indications the remote is connected. 

The ESP32 believes its connected (is.connected is true) 
The pi believes the remote is connected (shows up in BT menu as connected)

Battery % is updated on the Pi, so that GATT portion works.

However, the pi doesn't process any HID input from the ESP32
When I check the dmesg log on the pi I see this event over and over again. 
`Bluetooth: hci1: unexpected event for opcode 0x2016`

It's difficult to reproduce, but I'm not sure what opcode 0x2016 is. I can try to provide any specific logs that are needed.

drdobbins, 2021-07-18:
Looks like this issue might be on the raspberry pi side. 

I was able to get a system in the "bad state" and when running btmon I can see the keyboard events coming in.

`> ACL Data RX: Handle 64 flags 0x02 dlen 15                   [hci0] 659.714638
      ATT: Handle Value Notification (0x1b) len 10
        Handle: 0x0041
          Data: 0000060000000000
> ACL Data RX: Handle 64 flags 0x02 dlen 15                   [hci0] 659.715401
      ATT: Handle Value Notification (0x1b) len 10
        Handle: 0x0041
          Data: 0000000000000000
`

battery info seems to be accepted and updated as well. Power cycling the esp32 doesn't seem to help. Only reboot the pi will clear this.

T-vK, 2021-07-18:
Honestly, it wouldn't really surprise me if this actually is a problem with the Pi. I remember having all sorts of issues with Bluetooth mice and keyboards (that were not ESP32-based) with the Pi.

Things you could try are: Unpairing and disconnecting before restarting the ESP32. And using a shorter device name. Apple devices for example are known for having issues with ESP32 devices that use the default name I specified which is think is "ESP32 BLE Keyboard". Maybe the Pi has a similar issue.

drdobbins, 2021-07-18:
I'll give that a shot. Right now my devices are named "XXX 012" 

For some reason on the Pi side I can't pair the esp32 to the pi without enabling the serial port profile, which seems unnecessary since it's a HID device. 

Ex: 
`sudo nano /etc/systemd/system/dbus-org.bluez.service
*add -C and serial port profile
ExecStart=/usr/lib/bluetooth/bluetoothd -C
ExecStartPost=/usr/bin/sdptool add SP
`